pikvm/pikvm #92: Raspberry PI 4 HDMI to CSI2 Adapter - USB Splitter -> kvmd-vnc.service issues

PaulG4H, 2020-10-14:
**Describe the bug**
today I have done the pacman update and reboot, now the Log shows tons of error messages (8726 lines per second!) like

[2020-10-14 19:28:02 kvmd-vnc.service] --- aiohttp.internal               WARNING --- Multiple access to StreamReader in eof state, might be infinite loop.
[2020-10-14 19:28:02 kvmd-vnc.service] --- Stack (most recent call last):
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/bin/kvmd-vnc", line 9, in <module>
[2020-10-14 19:28:02 kvmd-vnc.service] ---     main()
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/kvmd/apps/vnc/__init__.py", line 47, in main
[2020-10-14 19:28:02 kvmd-vnc.service] ---     VncServer(
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/kvmd/apps/vnc/server.py", line 431, in run
[2020-10-14 19:28:02 kvmd-vnc.service] ---     loop.run_forever()
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
[2020-10-14 19:28:02 kvmd-vnc.service] ---     self._run_once()
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/asyncio/base_events.py", line 1859, in _run_once
[2020-10-14 19:28:02 kvmd-vnc.service] ---     handle._run()
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/asyncio/events.py", line 81, in _run
[2020-10-14 19:28:02 kvmd-vnc.service] ---     self._context.run(self._callback, *self._args)
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/kvmd/apps/vnc/rfb/__init__.py", line 103, in __wrapper
[2020-10-14 19:28:02 kvmd-vnc.service] ---     await coro
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/kvmd/apps/vnc/server.py", line 184, in __streamer_task_loop
[2020-10-14 19:28:02 kvmd-vnc.service] ---     async for (online, width, height, jpeg) in self.__streamer.read_stream():
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/kvmd/clients/streamer.py", line 73, in read_stream
[2020-10-14 19:28:02 kvmd-vnc.service] ---     data = bytes(await frame.read())
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/aiohttp/multipart.py", line 288, in read
[2020-10-14 19:28:02 kvmd-vnc.service] ---     data.extend((await self.read_chunk(self.chunk_size)))
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/aiohttp/multipart.py", line 301, in read_chunk
[2020-10-14 19:28:02 kvmd-vnc.service] ---     chunk = await self._read_chunk_from_length(size)
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/aiohttp/multipart.py", line 320, in _read_chunk_from_length
[2020-10-14 19:28:02 kvmd-vnc.service] ---     chunk = await self._content.read(chunk_size)
[2020-10-14 19:28:02 kvmd-vnc.service] ---   File "/usr/lib/python3.8/site-packages/aiohttp/streams.py", line 344, in read
[2020-10-14 19:28:02 kvmd-vnc.service] ---     internal_logger.warning(

mdevaev, 2020-10-14:
Please provide KVMD version: `pacman -Q | grep kvmd`.

mdevaev, 2020-10-14:
I'm sorry, I didn't notice that you were updated. Okay, I'll see what's going on. Simply restart VNC server: `systemctl restart kvmd-vnc`.

PaulG4H, 2020-10-14:
Hello mdevaev,

first wow I am surprised about your project and can imagine how much work you invest into.

After I stop the pikvm-vnc service the logging of the errors stop, then after the next reboot the log flooding stops.

PaulG4H, 2020-10-14:
Hello mdevaev,

mdevaev, 2020-10-17:
I'm going to reopn this because I want to deal with stacktrace.

mdevaev, 2020-10-17:
Okay, I think I fixed it.

closed 2020-10-17 by mdevaev