pikvm/pikvm #93: Raspberry PI 4 - HDMI to CSI2 - USB V3 --> Mouse by Webinterface and also VNC not usable

PaulG4H, 2020-10-14:
I connect an other Raspberry PI 3 to the HDMI CSI Adapter and the USB Cable, but the mouse latency is around 5 to 20 seconds, so not usable by Webinterface and VNC Client too.

Even changing screen resolution and frame rate or quality not reduces this effect.

But when I connect an Mouse localy to the Raspberry PI 3 the KVM shows the Mouse with an usable latency so this is not screen related.

But the mounting of an iso as CD-Rom works well.

Maybe you are so kind and point me to the direction how I am able to fix this?

Best regards
Paul

mdevaev, 2020-10-14:
So, RPi4 as KVM and RPi3 as server?

PaulG4H, 2020-10-14:
Yes, RPI4 as KVM and RPI3 as Server with an current Buster Full Rasberry OS

mdevaev, 2020-10-14:
Show `journalctl -u kvmd`.

PaulG4H, 2020-10-14:
-- Logs begin at Sat 2020-10-03 21:13:45 MSK, end at Wed 2020-10-14 22:04:00 MSK. --
Oct 03 21:13:51 pikvm systemd[1]: Started Pi-KVM - The main daemon.
Oct 03 21:13:53 pikvm kvmd[347]: kvmd.apps.kvmd                    INFO --- Using internal auth service 'htpasswd'
Oct 03 21:13:53 pikvm kvmd[347]: kvmd.plugins.msd.otg              INFO --- Using OTG gadget 'kvmd' as MSD
Oct 03 21:13:53 pikvm kvmd[347]: kvmd.plugins.msd.otg              INFO --- Probing to remount storage ...
Oct 03 21:13:53 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Remounting internal storage to RW ...
Oct 03 21:13:53 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-helper-otgmsd-remount', 'rw'] ...
Oct 03 21:13:54 pikvm sudo[359]: pam_systemd_home(sudo:account): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Oct 03 21:13:54 pikvm sudo[359]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-remount rw
Oct 03 21:13:54 pikvm sudo[359]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Oct 03 21:13:54 pikvm sudo[359]: pam_unix(sudo:session): session closed for user root
Oct 03 21:13:54 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Remouning /var/lib/kvmd/msd to RW-mode ...
Oct 03 21:13:54 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Console: CHOWN --- kvmd - /var/lib/kvmd/msd/images
Oct 03 21:13:54 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Console: CHOWN --- kvmd - /var/lib/kvmd/msd/meta
Oct 03 21:13:54 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Remounting internal storage to RO ...
Oct 03 21:13:54 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-helper-otgmsd-remount', 'ro'] ...
Oct 03 21:13:54 pikvm sudo[370]: pam_systemd_home(sudo:account): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Oct 03 21:13:54 pikvm sudo[370]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-remount ro
Oct 03 21:13:54 pikvm sudo[370]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Oct 03 21:13:54 pikvm sudo[370]: pam_unix(sudo:session): session closed for user root
Oct 03 21:13:54 pikvm kvmd[347]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Remouning /var/lib/kvmd/msd to RO-mode ...
Oct 03 21:13:54 pikvm kvmd[347]: kvmd.keyboard.keysym              INFO --- Reading keyboard layout /usr/share/kvmd/keymaps/en-us ...
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.apps.kvmd.server             INFO --- Preparing User-GPIO drivers ...
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.plugins.hid.otg              INFO --- Using UDC fe980000.usb
Oct 03 21:13:55 pikvm kvmd[381]: kvmd.plugins.hid.otg.device       INFO --- Started HID-keyboard pid=381
Oct 03 21:13:55 pikvm kvmd[382]: kvmd.plugins.hid.otg.device       INFO --- Started HID-mouse pid=382
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.apps.kvmd.ugpio              INFO --- Running User-GPIO drivers ...
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.plugins.msd.otg              INFO --- Watching for /var/lib/kvmd/msd/images
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.plugins.msd.otg              INFO --- Watching for /var/lib/kvmd/msd/meta
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.plugins.msd.otg              INFO --- Watching for /sys/kernel/config/usb_gadget/kvmd/functions/mass_storage.usb0/lun.0
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.apps.kvmd.streamer           INFO --- Installing SIGUSR2 streamer handler ...
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.apps.kvmd.http               INFO --- ======== Running on http://unix:/run/kvmd/kvmd.sock: ========
Oct 03 21:13:55 pikvm kvmd[347]: kvmd.apps.kvmd.http               INFO --- (Press CTRL+C to quit)
Oct 14 21:14:56 pikvm kvmd[347]: aiohttp.access                    INFO --- [- / -] 'GET /auth/check HTTP/1.0' => 401; size=284 --- referer='-'; user_agent='Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.11 Safari/537.36'
Oct 14 21:14:59 pikvm kvmd[347]: kvmd.apps.kvmd.auth               INFO --- Authorized user 'admin' via auth service 'htpasswd'
Oct 14 21:14:59 pikvm kvmd[347]: kvmd.apps.kvmd.api.auth           INFO --- Logged in user 'admin'
Oct 14 21:14:59 pikvm kvmd[347]: aiohttp.access                    INFO --- [- / 192.168.200.86] 'POST /auth/login HTTP/1.0' => 200; size=290 --- referer='https://192.168.199.44/login/'; user_agent='Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.11 Safari/537.36'
Oct 14 21:14:59 pikvm kvmd[347]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.199.44/login/'; user_agent='Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.11 Safari/537.36'
Oct 14 21:14:59 pikvm kvmd[347]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.199.44/login/'; user_agent='Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.11 Safari/537.36'
Oct 14 21:14:59 pikvm kvmd[347]: aiohttp.access                    INFO --- [admin (token) / 192.168.200.86] 'GET /info?fields=meta,extras HTTP/1.0' => 200; size=1489 --- referer='https://192.168.199.44 [... cut ...]
Oct 14 21:15:05 pikvm kvmd[347]: kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [193.081    stream] -- Can't switch OMX component state to OMX_StateExecuting, need to retry: OMX_ErrorInsufficientResources
Oct 14 21:15:05 pikvm kvmd[347]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [193.092    stream] -- Using JPEG quality: 80%
Oct 14 21:15:05 pikvm kvmd[347]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [193.092    stream] -- Creating pool with 3 workers ...
Oct 14 21:15:05 pikvm kvmd[347]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [193.092    stream] -- Capturing

mdevaev, 2020-10-14:
Try to add `usbhid.mousepoll=0` to /boot/cmdline.txt on RPi**3** and reboot.

mdevaev, 2020-10-14:
BTW is there a lag when using the keyboard?

PaulG4H, 2020-10-14:
No lag when use the keyboard, just the mouse, let my try your suggestion...

PaulG4H, 2020-10-14:
That was it, now the lateny is just some millimetres... 

Maybe you can create some sort of wiki where we can find such tips and do not need to bother you directly.

Great work, I will present your solution maybe we need some V3 hats in the future!

closed 2020-10-14 by PaulG4H

mdevaev, 2020-10-14:
Cool! Glad to hear it :)

Also added this problem to the documentation: https://github.com/pikvm/pikvm/blob/master/pages/troubleshooting.md#big-mouse-latency-on-rpi-as-managed-server