pikvm/pikvm #97: Transport from audio?

gitpower2017, 2020-10-20:
there is the possibility that audio is also transmitted.
Maybe the RPI will report to the PC/server as a USB soundcard?

gitpower2017, 2020-10-20:
sorry for doppel posting. Ther was an error on my Browser

mdevaev, 2020-10-20:
I will do it. There are two USB and TC358743 capabilities. Both will be supported.

torpak, 2020-11-25:
@mdevaev TC358743 would mean getting the hdmi audio channel?

piezza, 2020-12-17:
Would it also be possible to capture sound? My usecase is to use tools like MS Teams over pikvm remotely. So playing back and capturing sound in browser window would be cool!

mdevaev, 2020-12-17:
Not right now but I'm working on it.

rtrombetta, 2021-05-03:
There´s a way to do bidirectional audio ?

mdevaev, 2021-05-03:
Not yet

mdevaev, 2021-08-02:
Just some links for me to not forget:
- https://tools.ietf.org/id/draft-nandakumar-rtcweb-sdp-01.html
- https://tools.ietf.org/id/draft-ietf-rtcweb-sdp-08.html#rfc.section.5.2.6
- https://datatracker.ietf.org/doc/html/rfc4566
- https://datatracker.ietf.org/doc/html/draft-ietf-mmusic-data-channel-sdpneg-25
- https://developer.mozilla.org/ru/docs/Web/Media/Formats/WebRTC_codecs
- https://gist.github.com/albanpeignier/104902
- https://github.com/xiph/vorbis/blob/master/examples/encoder_example.c
- https://www.uni-due.de/~be0001/rn/index_e.html

GralfR, 2022-01-11:
I'm just getting into it with what's laying around (pi4 2GB with USB-HDMI-capture).
The first boot was good. Got my picture. And was asking for sound/audio, too.
I got my test-signal from a DVB receiver/settopbox. That could be a usecase, too, but needs audio.

Looking forward to it.

chilman408, 2022-02-07:
Question:  When audio support is enabled, will it be enabled on V2 as well or just V3?  

If it will be enabled for V2, is the current recommended hardware applicable (like the geekworm HDMI-CSI2)?

mdevaev, 2022-02-08:
Geekworm have some hardware bugs so audio will not work on it.

chilman408, 2022-03-08:
> Geekworm have some hardware bugs so audio will not work on it.

Thank you, do you have a recommended hardware setup for V2?  Would the Auvidea B101 work?

mdevaev, 2022-03-08:
B101 will work, yes.

chilman408, 2022-03-21:
Sorry one question, will the Auvidea B102 work?

mdevaev, 2022-03-21:
Yes

awkspace, 2022-05-03:
Would it be possible to offer a bounty for audio support over HDMI? This would be useful to my workflow and I'd be happy to donate towards it being done.

awkspace, 2022-05-03:
@mdevaev My understanding is that the hardware is available, but not the software, correct? I own a pre-assembled PiKVM V3 from PiShop.us, but there is no option to stream the audio from the WebUI. My question was whether I could donate to the development of the software for this.

mdevaev, 2022-05-03:
This feature is implemented in the hardware, yes. You can donate here if you want, https://patreon.com/pikvm, I will be grateful :)

piezza, 2022-05-04:
I just wonder if it is not also possible to use Linux USB Audio Gadget driver for audio and transfer it bi-directional over WebRTC. Having both, speaker and mic would be great. I assume over HDMI only one direction will work?

mdevaev, 2022-05-04:
Yep, only one direction.

mdevaev, 2022-05-26:
The reason is software only.

mdevaev, 2022-06-08:
V3 users can test the audio right now: https://docs.pikvm.org/audio/

bora-fsd, 2022-06-08:
Thank you @mdevaev. To confirm, is this to receive audio from the pikvm connected machine only? Or is there support to transmit microphone from web browser to pikvm as well?

mdevaev, 2022-06-08:
This is only to recieve audio from the machine. Microphone support is not yet in the plans.

json81, 2022-06-10:
> V3 users can test the audio right now: https://docs.pikvm.org/audio/

Tested and working!

awkspace, 2022-06-16:
Another 👍 here. Audio is working great on my v3. Thank you so much!!

mdevaev, 2022-06-28:
I suppose it can be considered solved now :)

closed 2022-06-28 by mdevaev

moezden, 2024-09-14:
Hi @mdevaev, regarding 

> "Reverse audio transmission, that is, from the browser towards PiKVM (for example, to emulate a microphone) is not yet supported."

Can we expect a feature for mic emulation soon?

mdevaev, 2024-09-14:
No, sorry.

mdevaev, 2025-01-20:
Official [PiKVM V3 and V4 devices](https://pikvm.org/buy) gets two-way audio with microphone.

Now you can use voice applications remotely, or you can just emotionally tell the server what you think about it.

**https://docs.pikvm.org/audio**

*PS: I know that many people wanted this feature. I spent all the New Year holidays on it, so if you found PiKVM useful and finally got what you wanted, you can support my work on PiKVM or just give me a beer with donation to [Patreon](https://www.patreon.com/pikvm) or [Paypal](https://paypal.me/pikvm). Happy PiKVMing!*