pikvm/pikvm #100: NFS or SMB Source for Virtual Media

zer010gic, 2020-10-21:
I would love to see the ability to have an NFS source or SMB source for virtual media. 

It would be nice to be able to set this both in the GUI as well as in a config file.

mdevaev, 2021-08-01:
Realted: https://www.raspberrypi.org/forums/viewtopic.php?t=267231

zer010gic, 2021-08-01:
I think that link is slightly different then my intent.  In that link they are describing using a NFS mount and presenting that via OTG MSD as a virtual drive.  What I am requesting is using NFS to present to the Pi an repository of ISO images that can be added to the list of usable images.  I srepac on the discord is doing something similar using symlinks to /var/lib/kvmd/msd/images.

mdevaev, 2021-08-01:
Is it working with symlinks?

ebartz, 2022-01-08:
I just did this with the root user.

First install nfs clients

```shell
rw
pacman -Sy --noconfirm nfs-utils
```

Them make sure to add a new mount point to the `/etc/fstab` like this:

```
192.168.0.123:/mnt/ISO    /var/lib/kvmd/msd/images    nfs    defaults    0 0 
```

Afterwards do a reboot and the system will now use nfs to store the images.

Would be great to configure this in the UI but this also works for me.

Rainee4563, 2022-05-12:
any update on this I have tried to do this manually and I'm getting errors with msd not loading

mdevaev, 2023-03-21:
It's done: https://docs.pikvm.org/msd/#nfs-storage

closed 2023-03-21 by mdevaev

LordHumphrey, 2023-09-22:
> It's done: https://docs.pikvm.org/msd/#nfs-storage完成：https://docs.pikvm.org/msd/#nfs-storage

Whether SMB is supported?

mdevaev, 2023-09-23:
I didn't test it, but it should. You will have to figure out how to set timeouts as for NFS.

LordHumphrey, 2023-09-25:
[image]

I have mount SMB on `/var/lib/kvmd`, and restart the kvmd, the log say `RuntimeError: Can't find 'otgmsd' mountpoint`.

Here is my `override.yaml`.
```yaml
kvmd:
    streamer:  
        max_fps: 30
```
I have tried NFS and webdav, got the same result.

Last, I'm not the english speaker, maybe some grammer mistake in my message. I'm so sorry that.

mdevaev, 2023-09-25:
> RuntimeError: Can't find 'otgmsd' mountpoint.

You should apply the X-kvmd mount parameters in fstab for the new partition. See how it was done with the default settings.