pikvm/pikvm #101: Streaming virtual disc images

kiler129, 2020-10-25:
**Is your feature request related to a problem? Please describe.**
One of the best features I use on systems like iLO is the ability to just mount an ISO image and reinstall the server. While booting an image from the PiKVM host is supported this solution poses problems when networks are configured to be secure.

Usually it's recommended that the management network is separated, at least on the switch level with a dedicated port. Currently PiKVM is a perfect solution for that. However, usually management networks are isolated and with no direct access to the Internet.

This configuration forces me to upload the image to PiKVM host through a VPN to mount it. It's a mild inconvenience.

**Describe the solution you'd like**
There should be an option to stream the ISO image through the browser:

[image]
[image]

**Describe alternatives you've considered**
Uploading the image to PiKVM & mounting it.

**Additional context**
- Additionally, making PiKVM datastore read-only could make it more reliable for long term use. With the requirement of r/w storage for images the fs cannot be made read-only.
- iLO serves ISO images via WebSocket. If there will be an interest I can poke around for more details.

WDYT?

mdevaev, 2020-10-25:
This is a good idea, but at the moment it is associated with some difficulties, local mounting of images is limitation of the kernel. However, it would be nice to do it somehow.

BTW the Pi-KVM file system is read-only. Images are uploaded on a separate partition (not rootfs) that is also permanently read-only and only remounted to read-write while uploading.

kiler129, 2020-10-25:
> This is a good idea, but at the moment it is associated with some difficulties, local mounting of images is limitation of the kernel. 

Do you know about any limitations as of know, or it just wasn’t expired previously?

I’m asking since I can poke around but if someone already tried and stepped into some issues it will be nice to not repeat the same steps :)

mdevaev, 2020-10-25:
No, no one has tried it. The kernel just requires specifying a local image, that's all. Maybe we can create a fuse driver and it will swallow it, but I haven't tried it.

mdevaev, 2021-07-16:
some information so as not to forget: https://wicg.github.io/file-system-access

mdevaev, 2026-10-02:
We have a first implementation of this on the Redfish API level: #830 For now we have support of streaming via HTTP(S), SMB and SFTP. I'll do browser streaming as the next step.