pikvm/pikvm #105: Shopping list for wired ethernet pikvm?

marmalodak, 2020-11-05:
Being unfamiliar with the Raspberry Pi ecosystem, I need a little help with putting together a complete build.

My preferences:

1. Wired ethernet
2. Display that will show me the IP address of the PIKVM
3. Plugs and power supplies so that I can plug it into the wall permanently
4. Case that can contain everything

This is what I have so far:

[Raspberry Pi 4 Model b](https://www.raspberrypi.org/products/raspberry-pi-4-model-b/)
[Raspberry Pi 4 case](https://www.raspberrypi.org/products/raspberry-pi-4-case/)
[Lusya Standard HDMI to CSI-2 Adapter Board](https://www.aliexpress.com/item/4000102166176.html)
[240×320, General 2inch IPS LCD Display Module](https://www.pishop.us/product/240-320-general-2inch-ips-lcd-display-module/)

Questions:

1. Can an ordinary macbook USB-C wall plug power the whole PIKVM device? 
2. Are there any parts I'm missing or would be a better replacement?
3. If I don't need a touch screen, which display might you recommend?
4. Is this [Raspberry Pi HDMI to CSI-2 Adapter Board with 15 pin FFC cable](https://geekworm.com/products/raspberry-pi-hdmi-to-csi-2-adapter-board-with-15-pin-ffc-cable) an appropriate capture device?
5. Is this [HDMI Adapter Board HDMI to CSI-2 TC358743XBG for Raspberry Pi 4B 3B 3B+ Zero](https://www.amazon.com/Adapter-Board-CSI-2-TC358743XBG-Raspberry/dp/B08D3H6NR4) an appropriate capture device? Why is it so much more expensive?

mdevaev, 2020-11-05:
> Can an ordinary macbook USB-C wall plug power the whole PIKVM device?
Perhaps no baceuse the Pi have a broken power delivery. I recommend the official RPi power supply.

> Are there any parts I'm missing or would be a better replacement?
USB-C wires and the power splitter.

> appropriate capture device?
Yep

> Why is it so much more expensive?
Becaus it's Zero compatible I guess.

marmalodak, 2020-11-06:
Thanks for the feedback.

Is the power splitter needed because the capture device and the screen each need their own power supplies? I don't see a product on the raspberry pi site that is an obvious power splitter. 

Is [this](https://www.amazon.com/Headphone-Charger-Splitter-Charging-Compatible/dp/B087N52WD2/ref=sr_1_3?dchild=1&keywords=usb-c+splitter&sr=8-3) an appropriate splitter?

If I get [this power supply](https://www.canakit.com/official-raspberry-pi-4-power-supply.html?cid=usd&src=raspberrypi), can I power both the RPi and the capture device with it?

I gather that I also need micro SD card. [Like this one maybe?](https://www.amazon.com/dp/B06XWN9Q99/ref=cm_sw_em_r_mt_dp_x0kPFb0N2D2EF)

mdevaev, 2020-11-06:
The splitter is required so that your server does not try to power the raspberry and power from the raspberry does not go back to the server.

> Is this an appropriate splitter?

No, you need a [separate power circuit](https://github.com/pikvm/pikvm/blob/master/img/v2_splitter.png). Like [this](https://www.tindie.com/products/8086net/usb-cpwr-splitter/)

> If I get this power supply, can I power both the RPi and the capture device with it?

The capture device does not need its own power supply. This power supply is a great choice.

> I gather that I also need micro SD card. Like this one maybe?

mdevaev, 2020-11-12:
I'm closing this issue, since apparently the problem is solved. If you encounter any difficulties, you can get support from discord :smirk_cat:

closed 2020-11-12 by mdevaev