pikvm/pikvm #107: Alternate SSH port & localhost access only build option

generalmanager, 2020-11-06:
I personally prefer to keep the attack surface of my devices to a minimum and a webserver is a rather big and especially jucy one on a KVM. Thus I like making sure such things are only exposed indirectly via a nonstandard port via SSH by using port forwarding.

Anyways it'd be nice to have an easy way to 
1. define the SSH port in the (build) config or the override config. As I haven't finished my hw build and couldn't easily find the default override config, so excuse me if that's already possible.
2. make the webserver only accessible from localhost during build time or via override option
3. have a short example and explanation of SSH port forwarding in the docs so less advanced users get a hint what to do if they'd like to improve on security with little effort. Defining the port forwarding in the .ssh/config file makes it even more convenient.

mdevaev, 2020-11-07:
Hi! Your request is clear and it's a good idea. However, I am currently busy with v3 and have to take the task of setting up the build environment to the background. I can offer you two ways:
1. You can edit the build Dockerfile to make the necessary changes to the image.
2. You can change the image after building.
In both cases, the changes will not be too complicated. This mainly applies to editing `/etc/kvmd/nginx.conf`

generalmanager, 2020-11-07:
Thanks for considering my ideas and pointing me at the right places. :)
I understand that you've got your hands full with the new HW build, there's no rush. I'm eagerly awaiting to get my hands on one of those. ;)
Glad to hear you'll take a look at it, once you've got some time to breathe again. :+1:

mdevaev, 2022-07-31:
A lot of time has passed, but in fact there is already a solution. I have changed the structure of Nginx configuration files so that you can easily edit the Dockerfile and change the ports when building the image:

```
[root@pikvm ~]# cat /etc/kvmd/nginx/listen-http.conf
listen 80;
listen [::]:80;
[root@pikvm ~]# cat /etc/kvmd/nginx/listen-https.conf
listen 443 ssl http2;
listen [::]:443 ssl http2;
```

Since the nginx configuration is too flexible, it seems impractical to make dedicated build variables for each unique case. So this is the best I can offer. I hope this helps.

Similarly, you can edit sshd_config: `sed -i -e "s/^\s*#*\s*Port\s\+.*$/Port $SSH_PORT/g" /etc/ssh/sshd_config`

closed 2022-07-31 by mdevaev