pikvm/pikvm #109: Failed to start Pi-KVM - EDID loader for TC358743

ndastur, 2020-11-07:
**Describe the bug**
I can open the KVM but I don't get a HMDI stream
The error message on console says to look at the status. That is pasted below.

```shell
# systemctl status kvmd-tc358743.service
* kvmd-tc358743.service - Pi-KVM - EDID loader for TC358743
     Loaded: loaded (/usr/lib/systemd/system/kvmd-tc358743.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2020-11-07 15:53:17 GMT; 4min 19s ago
    Process: 367 ExecStart=/usr/bin/v4l2-ctl --device=/dev/kvmd-video --set-edid=file=/etc/kvmd/tc358743-edid.hex --fix-edid-checksums (code=exited, status=1/FAILURE)
   Main PID: 367 (code=exited, status=1/FAILURE)

Nov 07 15:53:16 pikvm systemd[1]: Starting Pi-KVM - EDID loader for TC358743...
Nov 07 15:53:16 pikvm v4l2-ctl[367]: Cannot open device /dev/kvmd-video, exiting.
Nov 07 15:53:17 pikvm systemd[1]: kvmd-tc358743.service: Main process exited, code=exited, status=1/FAILURE
Nov 07 15:53:17 pikvm systemd[1]: kvmd-tc358743.service: Failed with result 'exit-code'.
Nov 07 15:53:17 pikvm systemd[1]: Failed to start Pi-KVM - EDID loader for TC358743.
```

**To Reproduce**
Steps to reproduce the behavior, like:
1. Burn brand new image
2. Connect to HMDI source
3. Turn on both
4. Goto to https server and click KVM
5. No stream

**Expected behavior**
A stream

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: Windows 10 OS as source. Pi Zero W
 - Browser Safari
 - Version From Git. Built own image as per docs

**Pi-KVM info:**
  - Raspberry Pi board version: Pi Zero W
  - Pi-KVM platform v2-hdmi
  - Video capture type HDMI board https://www.aliexpress.com/item/4000152180240.html?spm=a2g0s.9042311.0.0.66ca4c4dvnIGZi. Standard board with a Pi to Pi zero CSI connector
  - KVMD version:

```
kvmd 2.3-1
kvmd-platform-v2-hdmi-zerow 2.3-1
kvmd-webterm 0.35-1
```

  - uStreamer version: ```ustreamer 2.1-1```
  - Linux kernel: ```Linux pikvm 5.4.74-1-ARCH #1 SMP PREEMPT Wed Nov 4 16:52:07 UTC 2020 armv6l GNU/Linux```

mdevaev, 2020-11-07:
Try dmesg|grep tc3

ndastur, 2020-11-08:
```
# dmesg|grep tc3
[   25.911242] tc358743 10-000f: i2c_rd: reading register 0x0 from 0xf failed
[   25.911254] tc358743 10-000f: not a TC358743 on address 0x1e
```

mdevaev, 2020-11-08:
You probably connected the cable incorrectly. Look carefully at the location of the contacts, make sure that you have inserted the cable all the way and snapped the connector's latch.

ndastur, 2020-11-08:
How silly. Thank you. That's exactly right.

For anyone else. The connectors go into both devices facing downwards.

Next step is adding a pzem-004t power monitor.

mdevaev, 2020-11-08:
Great)

closed 2020-11-08 by mdevaev

Priyadharshini494, 2024-02-29:
I'M facing the same issue although i have connected the cables properly

allquixotic, 2024-03-31:
I'm facing the same issue on a stock PiKVM v4 Plus. I haven't opened or touched anything internal to the device - I've only ever used it from the "outside" (plugging things into the ports) and it still fails with this exact dmesg error. Any clues?