pikvm/pikvm #113: Clarify which HDMI to CSI-2 to buy

reedy, 2020-11-22:
Forking from https://github.com/pikvm/pikvm/pull/112 and a discussion on discord...

In https://github.com/pikvm/pikvm/blob/9e1a1ad/README.md#hardware-for-v2 it says

>Recommended: HDMI to CSI-2 bridge based on TC358743 - low latency ~100ms, more reliable.

Which links to https://aliexpress.com/item/4000102166176.html, but that page has two variants; `Color: Silver` which has two CSI ports and 2 cables, and `Color: Black` which has 1 CSI port and 1 CSI cable.. They are different sizes, and the "Silver" is the same size as a pi-zero, and the "Black" is slightly smaller

As people are designing RPI cases based on integrating these boards, I'm guessing most people buy one type or the other (presumably the silver? As it is the main one on the page... but I ended up buying the black...). Would be useful to clarify this and mention it in the readme!

[image]
[image]

mdevaev, 2020-11-22:
The section about selecting devices should be expanded. Due to the upcoming audio support, I need to check which devices it works on, and then improve the documentation.

db260179, 2020-12-29:
So I own both, the silver (long version) supports all raspberry pi's.

Does the black model work on the pi zero w?

mdevaev, 2020-12-29:
Yes bout you weel need the CSI cable with adapter

dossjj, 2021-01-20:
I purchased this CSI-2 module on Amazon but am not having luck getting a signal in the web portal. KVM showing No Signal.

https://www.amazon.com/gp/product/B0899L6ZXZ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Any ideas if this should work or not? I am using this on a pizerow.

kkellner, 2021-01-20:
@dossjj I bought the same one on Amazon on January 14, 2021.  I have it connected to a RPi4b and it worked with no effort using these instructions for creating the pi image: https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md

I'm assuming you swapped the cable the was connected by default (which is for the non-zero) with the pizero cable.  Try reconnecting the cable on both ends and verify its inserted the correct direction.

dossjj, 2021-01-20:
> @dossjj I bought the same one on Amazon on January 14, 2021. I have it connected to a RPi4b and it worked with no effort using these instructions for creating the pi image: [...]

Thanks for confirming that this one works. I did swap the cables for the smaller pizero cable, but I'll try reconnecting the cables and try again.

tj31moll, 2021-02-12:
https://www.amazon.com/dp/B08PP181DR/ref=cm_sw_r_cp_apa_fabc_CE7T7KH32YS55F4NM57G?psc=1

Has anyone tried this one? Not sure if it will work.

imascarenhas, 2021-02-20:
> https://www.amazon.com/dp/B08PP181DR/ref=cm_sw_r_cp_apa_fabc_CE7T7KH32YS55F4NM57G?psc=1
> 
> Has anyone tried this one? Not sure if it will work.

These devices are just to extend the connection to between the Pi and Pi Cam using an HDMI cable, essentially using it as a set of wires. It is missing the electronics to decode an HDMI signal which the more expensive adaptors discussed above do before sending it to the Pi.

mdevaev, 2021-07-04:
It seems that this has been fixed for a long time.

closed 2021-07-04 by mdevaev