pikvm/pikvm #115: Option to entirely disable non-HID drivers

june012006, 2020-11-23:
**Is your feature request related to a problem? Please describe.**

The system I'm connected with needs to not have mass storage devices connected.

**Describe the solution you'd like**
Either an option in the override.yml to entirely disable the mass storage drivers, and the cd emulation drivers, or documentation on how to remove them entirely.   The goal is that when the PI is plugged in, the system being controlled only recognizes the HID devices, and the monitor, with no other drivers being provided, or installed.

**Describe alternatives you've considered**
I have followed the instructions for disabling the mass storage service, and while in the KVM interface, there is indeed no option to mount anything, when plugged in to the device, it still installs the mass storage device, and the "Linux File-CD Gadget USB Device" drivers.

**Additional context**
Add any other context or screenshots about the feature request here.

mdevaev, 2020-11-23:
Sup. Did you try this: https://github.com/pikvm/pikvm/blob/master/pages/cookbook.md#disable-mass-storage-emulation

june012006, 2020-11-23:
I did that, and on the WebUI side, the mass storage features aren't showing, as it should be.   When I plug it into the computer though, in addition to the HID drivers, it installs the mass storage, and cd gadget drivers, which I don't want

My override.yaml
# Use this file to override system settings and other configs like this:
#
# kvmd:
#     server:
#         heartbeat: 5.0
#
# It will be applied AFTER other configs and "!include" directives and BEFORE validation.

    kvmd:
        msd:
            type:  disabled

mdevaev, 2020-11-23:
Remove extra spaces before the options. Then restart the KVM

june012006, 2020-11-25:
That was the issue, thanks!

closed 2020-11-25 by june012006