pikvm/pikvm #116: I'm using H2C-RPI-B01 device as HDMI-CSI bridge, but PIKVM doesn't seem to be able to communicate with it.

hradec, 2020-11-24:
**Describe the bug**
I'm using H2C-RPI-B01 device as HDMI-CSI bridge with a pizerow, but PIKVM doesn't seem to be able to communicate with it. 
the device is this one: https://www.aliexpress.com/item/4000152180240.html?spm=a2g0s.9042311.0.0.27424c4d2kwBtm

I get these on dmesg as well: 
```
[   24.160100] tc358743 10-000f: i2c_rd: reading register 0x0 from 0xf failed
[   24.285521] tc358743 10-000f: not a TC358743 on address 0x1e
```

I've tried `i2cdetect -y 1` and got the error: 
```
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
```

maybe the module for i2c is not loaded for some reason? 

I've created the image successfully using arch on kernel 5.9.9, using `make os`.  

**Pi-KVM info:**
  - Raspberry Pi board version: pizerow
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge - H2C-RPI-B01 - https://www.aliexpress.com/item/4000152180240.html?spm=a2g0s.9042311.0.0.27424c4d2kwBtm
  - KVMD version: kvmd 2.4-1 / kvmd-platform-v2-hdmi-zerow 2.4-1 / kvmd-webterm 0.35-1
  - uStreamer version: ustreamer 2.1-1
  - Linux kernel: Linux pikvm 5.4.79-1-ARCH #1 SMP PREEMPT Mon Nov 23 22:17:50 UTC 2020 armv6l GNU/Linux

mdevaev, 2020-11-24:
Make sure that you have inserted the correct side of the cable into both connectors. Attach a photo of both ends of the cable.

hradec, 2020-12-03:
I tried booth sides of the cable, and also switching the ends between the HDMI and PiZeroW, and still get the same error. 
[image]

But I think I've figured it out. I'm using one of those "usb dongle" boards for the PiZeroW:
[image]
so I can plug it directly into a usb port on the computer I wanna to control, and plug a short HDMI on the other end. It's a really neat solution! 

Turns out the power that comes from the USB2 on the computer is not enough to power the CSI-HDMI bridge board. That's why it was failing. 

As soon as I plugged a 2amp power supply directly in the usb micro connector of the PiZeroW, it started up correctly: 
[image]

So, a computer USB port powers up the PiZeroW (with the wifi) without issues, but it doesn't have enough power to power up the CSI-HDMI bridge I'm using.

mdevaev, 2020-12-03:
Lack of power can lead to such problems, which is why I show in the interface and README that a good power supply is required.

I guess this problem is solved.

mdevaev, 2020-12-04:
> Maybe another one would work? someone?

I don't think so. All bridges consume the same current.

closed 2020-12-25 by mdevaev