pikvm/pikvm #117: Use Respberry Pi 4B usb3.0 port for HDMI to USB input

halida, 2020-11-25:
Hi, I see the document about how to connect the HDMI to USB input,
But the USB port on rpi is 2.0, which is slow, Can I change to the USB3.0 port?

Thanks.

mdevaev, 2020-11-25:
This is not necessary, because there is enough bandwidth for the usb dongle. If you decide to do this, you will need a different udev rule. See `/etc/udev/rules.d/99-kvmd.rules`

halida, 2020-11-26:
Update:

the rules use /usr/bin/kvmd-udev-hdmiusb-check to find USB,

vi /usr/bin/kvmd-udev-hdmiusb-check

In rpi4 part, change 1-1.4:1.0 to 1-1.2:1.0 (USB 3.0 port next to net port LED)
The name can be find by running:
udevadm info -a -p $(udevadm info -q path -n /dev/video0)

Find the name with description as usb video device.

closed 2020-11-26 by halida