pikvm/pikvm #123: No signal with some HDMI sources using HDMI-CSI Bridge

Bob2345de, 2020-12-04:
Hi, am trying to use PIKVM to control on a system with just one USB C port available, no other usb or HDMI/DVI ports etc. I have a USB-C -->HDMI+USB Adapter (cheap chinesium) connected to that system, to which I am connecting a pi zerow with HDMI-CSI Bridge and USB cable.  

The PIKVM ui shows everything connected and keyboard and mouse work. The Display symbol at the top right is also green, however the screen shows "No Signal" in large text.

I've tried the video output from the USB-C HDMI adapter with a normal monitor and it works without issue, and also tried a PC Hdmi output into the HDMI CSI adapter and that appears properly on the PIKVM UI.

The pi is getting enough power, so that is not the issue, it seems that the HDMI stream from the USB-C HDMI adapter is somehow not compatible with the either the  HDMI-CSI Bridge or something in software. 

Any ideas on how I can try to isolate/fix the issue?

Thanks

mdevaev, 2020-12-04:
Hi. Show `journalctl -u kvmd`. And also see what parameters your usb-c Converter is trying to apply to the pikvm "monitor".

Bob2345de, 2020-12-05:
In the mean time I was able to make a config change in the USB-C’s firmware  to output 1024x768 instead of 1080p which was default, it now works fine. Will revert firmware and send you output from journalctl -u kvmd.

many thanks! PIKVM is great piece of work!

closed 2020-12-13 by Bob2345de

mdevaev, 2020-12-13:
Perfect!