pikvm/pikvm #125: Send custom keycodes to PiKVM?

victorhooi, 2020-12-06:
I would like to use PiKVM to remotely control some ChromeOS based devices.

These devices often have specific keys for things like "Search", or "Power".

I can get the keycodes for these - e.g. 0x98 is power.

However, is there any way to send custom keycodes to a device from PiKVM?

And do you know if sending these will still work when the device is powered off?

mdevaev, 2020-12-06:
Scancodes are not just numbers and you can't send an abstract key number. The fact is that the USB HID standard specifies different devices and character tables. A normal keyboard can have not one, but two, or each table is bound to its own HID descriptor. Roughly speaking, each handle is a separate virtual device, and a normal keyboard on your Desk can contain two or three such devices on a single wire. The problem is that many special keys are not in the "main" table, but in additional ones. To send these keys, you need to pre-configure the virtual device that implements them.

The power key is a special case and is located in the main table. You can send it and the operating system will receive this signal. However, this probably won't work for the BIOS and the computer turned off, because the USB composite device is a fairly complex specification. It depends on the motherboard. Also, maybe the [Arduino HID](https://github.com/pikvm/pikvm/blob/master/pages/arduino_hid.md) will work. However, I don't have the ability to test this right now.

mdevaev, 2020-12-06:
As for custom keycodes, kvmd validates them, so no. But it's easy to add them to the list if you're missing something. If these codes are from the default table, then I will do it in a couple of minutes.

mdevaev, 2020-12-17:
I believe I've answered your question

closed 2020-12-17 by mdevaev