pikvm/pikvm #128: KVM 4 port switches supported list add to docs

db260179, 2020-12-09:
**Is your feature request related to a problem? Please describe.**
Keep a documented list of 4 port KVM hdmi switches that work with this project [...]

**Describe the solution you'd like**
Maintain a list of these KVM switches so customers know what they can use.

https://a.aliexpress.com/_BSpS8t

This 4 port KVM switch is a good price and has adaptive EDID on the hdmi port, which means you can plug/unplug and it will show up again with know issues. Also the keyboard shortcuts work from the web interface - CTRL + CTRL then port number (1,2,3,4)

Be good to keep a list of these for future use and customers using your project.

mdevaev, 2020-12-10:
Good idea. Does this switch require an Arduino HID or works with OTG?

https://github.com/pikvm/pikvm/blob/master/pages/multiport.md#list-of-tested-kvms

db260179, 2020-12-10:
> Good idea. Does this switch require an Arduino HID or works with OTG?
> 
> https://github.com/pikvm/pikvm/blob/master/pages/multiport.md#list-of-tested-kvms

So i've updated my main quote above.

Im using this on a Raspberry pi zero W via the OTG.

This particular kvm switch has usb hub ports as well, so you can plug in usb keys etc.

mdevaev, 2020-12-17:
Added

hberntsen, 2021-01-29:
When I connect my Auvidea B101 to the output of this KVM switch and apply power, the raspberry does not boot (green led of the Auvidea also does not light up). At the moment I unplug the HDMI cable, the raspberry starts and the auvidea led turns on. I can then plug it back in again and everything is fine. So as long as there is no power outage it works fine :)

BernhardLinz, 2021-01-29:
> When I connect my Auvidea B101 to the output of this KVM switch and apply power, the raspberry does not boot (green led of the Auvidea also does not light up). At the moment I unplug the HDMI cable, the raspberry starts and the auvidea led turns on. I can then plug it back in again and everything is fine. So as long as there is no power outage it works fine :)
> 
> (I have the AM-KVM401K-V2.1 2020-09-19)

Same Problem her. Need to unplug HDMI or to turn of the KVM switch (which have a little Power Switch at the front). The green LEDs of the KVM switch will still shine but the raspberry start to boot

joBr99, 2021-01-29:
Same problem here, but with a different kvm switch

hberntsen, 2021-01-29:
My AM-KVM401K-V2.1  HDMI switch has an AX68002. I found the datasheet on the ASIX website and found the reset pin for the chip. I tried keeping the CPU in reset but that did not help :(. I did find a UART port with a baud rate of 19200. It prints
```
[Virtual HUB] VID = 0b95, PID = 6804
 System Informaiton:
  (v) USB Storage Device:Yes,PageSize=512
HDMI_CH4 
> DC_Port:0, VBUS Attached
> DC_Port:0, Suspend
> DC_Port:0, Reset
> DC_Port:0, Reset
> DC_Port:0, Suspend
HDMI_CH3 
HDMI_CH2 
> DC_Port:0, Reset
> DC_Port:0, Reset
> DC_Port:0, Suspend
HDMI_CH1 
```
It prints the HDMI output line when a button on the front panel is pressed. It also prints some info on the attached USB port. I couldn't find the protocol to send commands to the device.

Unfortunately they removed the text on the chips so I don't know which HDMI switch chips they used. Looks like there are 2 3:1 chips on the board.

ingo916, 2021-04-14:
I have the AIMOS 4 port hdmi switch and seems to be working great, how do I add the UI elements to control the KVM switch? pressing ctrl ctrl then the port number will change the display? how do i replicate this in the UI?

strausmann, 2021-04-28:
I see in the Discord Chat that some people would like to buy this HDMI Switch from Tendak:

https://www.amazon.com/Tendak-Keyboard-Printer-Switcher-Include/dp/B08B7XXBGZ/ref=sr_1_3?dchild=1&keywords=tendak+4x1+hdmi+kvm+switch&qid=1618875053&sr=8-3

The Switch has an external Controller connected via Micro USB to the HDMI Switch. I can change the HDMI Source with the four buttons on this controller.

How can I connect the Control Port now with my Raspberry Pi 4?

arch1mede, 2021-04-28:
@strausmann from my search, it was 2 others that were asking about it. That specific model needs to be controlled by gpio and requires much more work than an average person can do, I would recommend finding a kvm that supports hotkey switching and a usb printer port for the OTG. Physical kb switching is still needed though.

@ingo916 At this time its not supported, physical kb is still needed.

mdevaev, 2021-07-04:
I suppose this issue can be closed since we have a list of supported switches. Problems with them and other models should be solved in Discord or other dedicated issues among users who have them.

closed 2021-07-04 by mdevaev

rkkotnik, 2022-11-23:
I have PiKVm with raspberry 4 and i Love it! I have a plan to add another server to my homelab and i would like to also manage that over PIKVM.

I found some KVM switch:
https://www.aliexpress.com/item/1005003157653420.html?spm=a2g0o.productlist.0.0.541b1c37lASam0&algo_pvid=01494e58-5a9e-4014-92b3-8a5f596165c2&algo_exp_id=01494e58-5a9e-4014-92b3-8a5f596165c2-15&pdp_ext_f=%7B%22sku_id%22%3A%2212000030190733104%22%7D&pdp_npi=2%40dis%21EUR%2120.21%2112.33%21%21%21%21%21%402103255a16692032284283014ed4c2%2112000030190733104%21sea&curPageLogUid=jxTP0s5j9N2L

But IDK if that will work. I know you have a list of devices that work, but many are for too many devices and are expensive. + I saw something about back power, which I don't quite understand, what is up with that?

arch1mede, 2022-11-24:
Hi, yes as was explained on Discord, we have not tested this, its likely it will not work and have the same or more limitations as the 4 port. If you are planning on getting this, please let us know if it works or not.