pikvm/pikvm #131: No Stream with HDMI CSI Bridge

danmed, 2020-12-12:
Hi,

I've just moved from a USB dongle to the CSI Bridge : 

https://www.aliexpress.com/item/4000102166176.html?spm=a2g0s.9042311.0.0.72054c4dtcFZ0s

I'm not getting a video stream. The light is on, i'm up to date, the HDMI cable is plugged in.

I'm seeing the below in the logs : 

```
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.146    stream] -- Device fd=10 opened
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.146    stream] -- Using input channel: 0
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.146    stream] -- Using TV standard: DEFAULT
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.152    stream] -- Got new DV timings: resolution=1920x1080, pixclk=148500000
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.172    stream] -- Using resolution: 1920x1080
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.172    stream] -- Using pixelformat: UYVY
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.172    stream] -- Querying HW FPS changing is not supported
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.172    stream] -- Using IO method: MMAP
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.186    stream] -- Requested 5 device buffers, got 5
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [172.189    stream] -- Unable to start capturing: Invalid argument
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.192    stream] -- Device fd=10 closed
[2020-12-12 21:29:24 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [172.192    stream] -- Sleeping 1 seconds before new stream init ...

```

[image]

libram11, 2020-12-16:
just for testing try a different cable, and try a different source? Mine came up just fine but my source HDMI had a little wobble in the connector and didnt want to display initially.  Also not your exact model of hdmi card, mine came up no problem. Geekworm Raspberry Pi HDMI in Module

Bob2345de, 2020-12-18:
had a similar issue, was caused by my hdmi source outputing 1080p @60hz, which exceaded the max input specs of my hdmi-csi bridge board. try to reduce output to lower resolution and or frame rate to see if that solve the issue

mdevaev, 2020-12-26:
What's up?

closed 2021-01-08 by mdevaev