pikvm/pikvm #133: Unable to compile OS

lucabert, 2020-12-17:
I'm trying to compile the software for running on ZeroW.
This is my config.mk:

```
BOARD = zerow
PLATFORM = v2-hdmi
HOSTNAME = pikvm
LOCALE = de_DE
TIMEZONE = Europe/Berlin
ROOT_PASSWD = root
WEBUI_ADMIN_PASSWD = admin
IPMI_ADMIN_PASSWD = verystrongpassword
CARD = /dev/mmcblk0
WIFI_ESSID = "myssid"
WIFI_PASSWD = "veryveryverystrongpassword"
```

Then I started "make os". After a while make fails with these errors:

```
Step 22/87 : RUN echo "Server = $REPO_URL/\$arch/\$repo" > /etc/pacman.d/mirrorlist     && pacman-key --init    && pacman-key --populate archlinuxarm
 ---> Running in 0a37e7f2034a
gpg: /etc/pacman.d/gnupg/trustdb.gpg: trust-db erzeugt
gpg: keine ultimativ vertrauenswürdigen Schlüssel gefunden
gpg: starting migration from earlier GnuPG versions
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: error: GnuPG agent unusable. Please check that a GnuPG agent can be started.
gpg: migration aborted
gpg: starting migration from earlier GnuPG versions
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: error: GnuPG agent unusable. Please check that a GnuPG agent can be started.
gpg: migration aborted
gpg: Generating pacman keyring master key...
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: agent_genkey failed: Agent läuft nicht
gpg: key generation failed: Agent läuft nicht
gpg: Done
==> Aktualisiere Trust-Datenbank...
gpg: "Trust-DB"-Überprüfung nicht nötig
==> Füge Schlüssel aus archlinuxarm.gpg hinzu...
gpg: starting migration from earlier GnuPG versions
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: error: GnuPG agent unusable. Please check that a GnuPG agent can be started.
gpg: migration aborted
gpg: Schlüssel B55C5315DCD9EE1A wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel B55C5315DCD9EE1A wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: Schlüssel BC704E86B823CD25 wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel BC704E86B823CD25 wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: Schlüssel BF7EEF7A9C6B5765 wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel BF7EEF7A9C6B5765 wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: Schlüssel BF7EEF7A9C6B5765 wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel BC704E86B823CD25 wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel B55C5315DCD9EE1A wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel 77193F152BDBE6A6 wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel 77193F152BDBE6A6 wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: can't connect to the agent: IPC "connect" Aufruf fehlgeschlagen
gpg: Schlüssel B55C5315DCD9EE1A wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel BC704E86B823CD25 wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel BF7EEF7A9C6B5765 wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel 77193F152BDBE6A6 wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel BC704E86B823CD25 wurde 16088 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
gpg: Schlüssel 77193F152BDBE6A6 wurde 3329 Tage in der Zukunft erzeugt (Zeitreise oder Uhrenproblem)
==> Signiere die vertrauenswürdigen Schlüssel im Schlüsselbund lokal...
  -> Signiere Schlüssel 69DD6C8FD314223E14362848BF7EEF7A9C6B5765 lokal...
==> FEHLER: 69DD6C8FD314223E14362848BF7EEF7A9C6B5765 konnte nicht lokal signiert werden.
  -> Signiere Schlüssel 02922214DE8981D14DC2ACABBC704E86B823CD25 lokal...
==> FEHLER: 02922214DE8981D14DC2ACABBC704E86B823CD25 konnte nicht lokal signiert werden.
  -> Signiere Schlüssel 9D22B7BB678DC056B1F7723CB55C5315DCD9EE1A lokal...
==> FEHLER: 9D22B7BB678DC056B1F7723CB55C5315DCD9EE1A konnte nicht lokal signiert werden.
The command '/bin/sh -c echo "Server = $REPO_URL/\$arch/\$repo" > /etc/pacman.d/mirrorlist      && pacman-key --init    && pacman-key --populate archlinuxarm' returned a non-zero code: 1
make[1]: *** [os] Error 1
make[1]: Leaving directory `/root/os/.pi-builder'
make: *** [os] Error 2
```

What am I doing wrong?
I'm trying to compile all on a CentOS 7 server.

Thanks
Luca

mdevaev, 2020-12-17:
Set the english locale plz, I don't understand.

lucabert, 2020-12-17:
Well, there is nothing more to see in english... ;)

```
Step 22/87 : RUN echo "Server = $REPO_URL/\$arch/\$repo" > /etc/pacman.d/mirrorlist     && pacman-key --init    && pacman-key --populate archlinuxarm
 ---> Running in cc6fba1e7718
gpg: /etc/pacman.d/gnupg/trustdb.gpg: trustdb created
gpg: no ultimately trusted keys found
gpg: starting migration from earlier GnuPG versions
gpg: can't connect to the agent: IPC connect call failed
gpg: error: GnuPG agent unusable. Please check that a GnuPG agent can be started.
gpg: migration aborted
gpg: starting migration from earlier GnuPG versions
gpg: can't connect to the agent: IPC connect call failed
gpg: error: GnuPG agent unusable. Please check that a GnuPG agent can be started.
gpg: migration aborted
gpg: Generating pacman keyring master key...
gpg: can't connect to the agent: IPC connect call failed
gpg: agent_genkey failed: No agent running
gpg: key generation failed: No agent running
gpg: Done
==> Updating trust database...
gpg: no need for a trustdb check
==> Appending keys from archlinuxarm.gpg...
gpg: starting migration from earlier GnuPG versions
gpg: can't connect to the agent: IPC connect call failed
gpg: error: GnuPG agent unusable. Please check that a GnuPG agent can be started.
gpg: migration aborted
gpg: key B55C5315DCD9EE1A was created 16088 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A was created 7716 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A was created 7716 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A was created 7716 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A: no valid user IDs
gpg: key BC704E86B823CD25 was created 16088 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25 was created 7716 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25 was created 7716 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25 was created 7716 days in the future (time warp or clock problem)
gpg: key BC704E86B823CD25: no valid user IDs
gpg: key BF7EEF7A9C6B5765 was created 16088 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765 was created 7716 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765 was created 7716 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765 was created 7716 days in the future (time warp or clock problem)
gpg: key BF7EEF7A9C6B5765: no valid user IDs
gpg: key 77193F152BDBE6A6 was created 16088 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6 was created 7716 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6 was created 7716 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6 was created 7716 days in the future (time warp or clock problem)
gpg: key 77193F152BDBE6A6: no valid user IDs
==> Locally signing trusted keys in keyring...
  -> Locally signing key 69DD6C8FD314223E14362848BF7EEF7A9C6B5765...
==> ERROR: 69DD6C8FD314223E14362848BF7EEF7A9C6B5765 could not be locally signed.
  -> Locally signing key 02922214DE8981D14DC2ACABBC704E86B823CD25...
==> ERROR: 02922214DE8981D14DC2ACABBC704E86B823CD25 could not be locally signed.
  -> Locally signing key 9D22B7BB678DC056B1F7723CB55C5315DCD9EE1A...
==> ERROR: 9D22B7BB678DC056B1F7723CB55C5315DCD9EE1A could not be locally signed.
The command '/bin/sh -c echo "Server = $REPO_URL/\$arch/\$repo" > /etc/pacman.d/mirrorlist      && pacman-key --init    && pacman-key --populate archlinuxarm' returned a non-zero code: 1
make[1]: *** [os] Error 1
make[1]: Leaving directory `/root/os/.pi-builder'
make: *** [os] Error 2
```

Thanks
Luca

mdevaev, 2020-12-17:
On older kernels, there is a problem of incorrect emulation of some system calls. Use ubuntu 20.04 in a virtual machine.

lucabert, 2020-12-17:
Oh, must I install a VM with Ubuntu just for that?!? I think Docker should manage these problems...

Anyway, nothing to do... I try installing a fresh Ubuntu 20.04 with docker&Co

If I'll still have problems, I'll comment this issue again.

Thanks
Luca

mdevaev, 2020-12-17:
The build is done by emulating ARM via QEMU. If your kernel is too old, it has problems that cause QEMU to work incorrectly. You need a new kernel on the host machine for the build to work.

lucabert, 2020-12-17:
OK, could you define "new kernel"? Starting from which version?

Thanks
Luca

mdevaev, 2020-12-17:
5.4 is working.

mdevaev, 2020-12-26:
It seems the problem is solved. Feel free to reopen if necessary.

closed 2020-12-26 by mdevaev