pikvm/pikvm #134: Problems with wifi in KVM mode

al-pankov, 2020-12-18:
**Describe the bug**
Pikvm is connected via wifi. After the system starts, the connection is stable, but as soon as you enter KVM mode and video streaming starts, packet losses begin about 90%, which makes it difficult to work with the device. Wifi operates in the 2.4 GHz band. From my observation, this started after switching to python 3.9. Additionally, I am using the wireguard kernel module to connect to remote resources, but I do not know how much this may affect the problem.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect to wifi
2. Click on 'KVM'
3. Packet looses begins

**Expected behavior**
Stable work via wifi.

**Screenshots**
https://prnt.sc/w5eoz7
https://prnt.sc/w5er8c

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 87.0.4280.88
 - VNC client: -

**Pi-KVM info:**
  - Raspberry Pi board version: RPi 4 2Gb
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: 2.8-1
  - uStreamer version: 2.2-1
  - Linux kernel: Linux pikvm 5.4.83-1-ARCH #1 SMP PREEMPT Mon Dec 14 15:43:54 UTC 2020 armv7l GNU/Linux

**Additional context**

mdevaev, 2020-12-18:
Python can't affect this in any way. The problem is probably related to kernel or firmware updates. Look to `dmesg`.

al-pankov, 2020-12-18:
Dmesg is clean, there are no errors at these moments.
I'll try to flash latest image and try again in Monday.

al-pankov, 2020-12-21:
I've installed a fresh system image, updated to the latest version and the problem has gone. 
False alarm.

closed 2020-12-21 by al-pankov