pikvm/pikvm #136: VNC server error

flexi-sys, 2020-12-22:
When connecting to the VNC server with TigerVNC connection is made, the screen is shown, then the connection is cut almost immediately.  

Output from systemctl status lvmd-vcn.service:

```
 * kvmd-vnc.service - Pi-KVM - VNC to KVMD/Streamer proxy
     Loaded: loaded (/usr/lib/systemd/system/kvmd-vnc.service; enabled; vendor preset: disabled)
     Active: active (running) since Mon 2020-12-21 18:36:05 GMT; 51min ago
   Main PID: 4320 (kvmd-vnc)
      Tasks: 3 (limit: 844)
     CGroup: /system.slice/kvmd-vnc.service
             `-4320 /usr/sbin/python /usr/bin/kvmd-vnc --run

Dec 21 19:27:47 pikvm kvmd-vnc[4320]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/vnc/rfb/__init__.py", line 103, in __wrapper
Dec 21 19:27:47 pikvm kvmd-vnc[4320]:     await coro
Dec 21 19:27:47 pikvm kvmd-vnc[4320]:   File "/usr/lib/python3.9/site-packages/kvmd/apps/vnc/server.py", line 151, in __kvmd_task_loop
Dec 21 19:27:47 pikvm kvmd-vnc[4320]:     async for event in self.__kvmd_ws.communicate():
Dec 21 19:27:47 pikvm kvmd-vnc[4320]:   File "/usr/lib/python3.9/site-packages/kvmd/clients/kvmd.py", line 159, in communicate
Dec 21 19:27:47 pikvm kvmd-vnc[4320]:     raise RuntimeError(f"Unhandled WS message type: {msg!r}")
Dec 21 19:27:47 pikvm kvmd-vnc[4320]: RuntimeError: Unhandled WS message type: WSMessage(type=<WSMsgType.CLOSED: 257>, data=None, extra=None)
Dec 21 19:27:47 pikvm kvmd-vnc[4320]: kvmd.apps.vnc.rfb                 INFO --- [main] [::ffff:192.168.10.209]:62280: Cancelling subtask ...
Dec 21 19:27:47 pikvm kvmd-vnc[4320]: kvmd.apps.vnc.rfb                 INFO --- [streamer] [::ffff:192.168.10.209]:62280: Cancelling subtask ...
Dec 21 19:27:47 pikvm kvmd-vnc[4320]: kvmd.apps.vnc.rfb                 INFO --- [entry] [::ffff:192.168.10.209]:62280: Connection closed
```

**System info:**
- MacOS Big Sur 11.1
- TigerVNC 1.11.80

**Pi-KVM info:**
  - Pi Zero W
  - v2-hdmi
  - CSI bridge
  - KVMD version:
```
kvmd 2.8-1
kvmd-platform-v2-hdmi-zerow 2.8-1
kvmd-webterm 0.35-1
```
  - uStreamer version: `ustreamer 2.2-1`
  - Linux kernel: `Linux pikvm 5.4.83-1-ARCH #1 SMP PREEMPT Mon Dec 14 15:44:54 UTC 2020 armv6l GNU/Linux`

**Additional context**
Have also tried to connect to the server with TigerVNC and TightVNC on Windows VM and have the same issue with the same errors.  Not sure if this is a bug or an incorrect set up, but haven't been able to resolve the issue on my own.

mdevaev, 2020-12-22:
Show `journalctl -u kvmd`

flexi-sys, 2020-12-22:
Typically, having restarted both lvmd-vnc & kvmd, I am not having any issues and TigerVNC is now connecting and seems stable.

If I can reproduce the issue again in the future I will re-post.  I've attached the output of kvmd and kvmd-vnc as a .txt for info in any case.

[kvmd_log.txt](https://github.com/pikvm/pikvm/files/5731274/kvmd_log.txt)
[kvmd-vnc_log.txt](https://github.com/pikvm/pikvm/files/5731275/kvmd-vnc_log.txt)

mdevaev, 2021-03-07:
Is there any additional information about this? I've tried to reproduce this myself over and over again, but I don't understand in what situations it occurs.

mdevaev, 2021-03-07:
I can see what exactly breaks down but I don't understand what causes it and I don't want to treat the symptoms.

slavikf1, 2021-06-08:
I've encountered the same issue on RP4B with HDMI-to-CSI. 
The only difference is that restarting does not help - the issue occurs every time with different VNC clients. No configuration other than initial VNC has been done.

mdevaev, 2021-06-08:
@slavikf1 perform system update, reboot and try again.

slavikf1, 2021-06-08:
Thanks.

Does not reproduce after the update!

mdevaev, 2021-07-14:
I believe that this problem is no longer relevant. The original bug was fixed, the other bug (if it still exists) is floating and is hiding somewhere in aiohttp. Now VNC uses my own library for video streaming instead of aiohttp, so the probability that the bug will occur again is close to zero. If someone encounters a similar problem again, write here, I will figure it out.

closed 2021-07-14 by mdevaev