pikvm/pikvm #137: No WiFi with v2 on Pi Zero W

lynix, 2020-12-22:
**Describe the bug**
I successfully managed to build an image from current _master_ for the v2 Pi Zero W platform with CSI Bridge. It boots up but sadly it does not connect to my WiFi network.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Follow image build guide in wiki
2. Flash image to Pi Zero W SD card
3. Power it on

**Expected behavior**
Pi Zero W connects to my WiFi network.

**Screenshots**
The only output that looks potentially relevant are these messages from the `brcmfmac` module (sorry for the crappy quality):
[image]

**Pi-KVM info:**
  - Raspberry Pi board version: RPi Zero W
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge

**Additional context**
It's quite hard to debug this issue as, on the Zero W, without WiFi there is no shell access. I might try to solder some pins for RS232. According to [raspberrypi/linux#3659](https://github.com/raspberrypi/linux/issues/3659) the message above is of cosmetic nature as there should be a clm blob baked into the firmware. However, my WiFi uses channel 13 on 2.4GHz (and channel 60 on 5GHz) so it may very well be a frequency issue.
I tried replacing the `netctl` stuff in the image with pure `systemd-networkd` and `wpa_supplicant` (launched via systemd), as I do on my other wireless boxes, but this didn't make any difference.

mdevaev, 2020-12-23:
Check the regular RPi OS from official website. If wifi does not work, then you got a defective board. It happens sometimes, I've come across it.

mdevaev, 2021-01-08:
What's up?

MichaelMichaelMichaelMichaelMichael, 2021-01-09:
I think I have the same issue with the self-built zerow image. "think" because I do not have a Mini-HDMI Connector so I cannot see the console output (I have the chicken egg problem, I need a PIKVM ;-) ). 
Would I have to escape any special characters in the WiFi password, like e.g. $, !, #, +, ...? At least one of them I had to escape upon build otherwise the build wouldn't go through. 
On the SD card, I do not see anything in the logs, it's empty, but the LED flickers, so it seems to boot up.

MichaelMichaelMichaelMichaelMichael, 2021-01-09:
[image][image]
[image]

I've managed to catch more error messages; those seem to be the same as OP's error messages.

MichaelMichaelMichaelMichaelMichael, 2021-01-09:
Got it running on the zerow ... the following steps were necessary for me to do it:

1. Escape the Wifi password so build doesn't fail
2. mount the SD card, edit `/etc/netctl/wlan0-WIFINAME` and take out the escape characters from the Wifi password
3. according to the tip on [this page](https://benjaminknofe.com/blog/2017/12/19/fix-broken-onboard-wifi-on-a-raspberry-pi-3-and-arch-linux-and-systemd/) I've written a little systemd service

```
[root@pikvm system]# pwd
/etc/systemd/system
[root@pikvm system]# cat network-prepare.service 
[Unit]
Description=Network preparations
Before=network-pre.target
Wants=network-pre.target

[Service]
ExecStart=lsmod
ExecStart=sleep 2
ExecStart=modprobe -r brcmfmac
ExecStart=sleep 2
ExecStart=modprobe brcmfmac
ExecStart=sleep 2
ExecStart=ifconfig -a
ExecStart=iwconfig 
```

I'm happy now.

mdevaev, 2021-01-09:
How did you get to this solution?

MichaelMichaelMichaelMichaelMichael, 2021-01-13:
See item #3 of my list above: 
"according to the tip on [this page](https://benjaminknofe.com/blog/2017/12/19/fix-broken-onboard-wifi-on-a-raspberry-pi-3-and-arch-linux-and-systemd/) I've written a little systemd service"

mdevaev, 2021-01-13:
Perfect. I've added this to the troubleshooting guide.

closed 2021-01-13 by mdevaev

lynix, 2021-03-20:
Well, looking at the systemd unit posted by @MichaelMichaelMichaelMichaelMichael I get the idea what it's trying to do. But I highly doubt it is actually doing anything, for multiple reasons:

Unless `Type=oneshot` is set, systemd expects exactly one `ExecStart`. The unit type in the example defaults to `simple` which makes each additional `ExecStart` overwrite the previous one, so in the end only `ExecStart=iwconfig` is even considered.

Systemd (iirc) requires absolute paths for binaries to be executed, so none of the `ExecStart` directives in the example would actually work.

`iwconfig` (as well as `lsmod` and `ifconfig -a`) would not do anything, I guess they are just there for trying to get some debugging output. The aim of the unit boils down to unloading and re-loading the `brcmfmac` module with a delay in between, which could be realized like this (only the service section of the unit given):

```
[Service]
Type=oneshot
ExecStart=/usr/bin/sh -c 'modprobe -r brcmfmac; sleep 2; modprobe brcmfmac'
RemainAfterExit=yes
```

To comment on my initial issue: **I solved it without touching the Pi at all**. It turned out to be the access point that was using channels not permitted by CRDA.