pikvm/pikvm #138: Arduino HID

xadonxander, 2020-12-22:
I can not seem to get the Arduino HID to work. Everything is set up like the pictures and the Arduino shows up as HID device on the PC. The keyboard and mouse icons at the top of the KVM page are red though. I tested all connections and all are good.

mdevaev, 2020-12-22:
If the flashing was successful, you may have mixed up the RX and TX wires.

xadonxander, 2020-12-23:
I thought the same thing. I swapped them. The HID reset is also not working

xadonxander, 2020-12-23:
I pulled the connectors off the raspberry pins and activated them manually. atx and reset work, somewhere in the programming the pi3 is not controlling the gpio correctly.

mdevaev, 2020-12-23:
Attach output of `journalctl -u kvmd`.

xadonxander, 2020-12-31:
I figured it out. The issue was my logic level converter. It needed +v on the enable pin. All the docs I read said it had a pullup resistor built in but it did not.

mdevaev, 2021-01-01:
Bueno

closed 2021-01-01 by mdevaev