pikvm/pikvm #139: Mass Storage devices not mounted properly

valeech, 2020-12-24:
When attaching any iso file the drive shows up in Windows but when I try to access the files I get and error that the disk structure is corrupted and unreadable.

I am able to successfully mount the iso directly on the pi so I know the iso is good. 

Everything else about the pikvm is functioning correctly. I am using a spiced USB cable to provide power to the pi 
and gadget access to the server. 

**Pi-KVM info:**
  - Raspberry Pi board: version 4
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge]
  - KVMD version: kvmd 1.100-1
  - uStreamer version: 1.25-1
  - Linux kernel: Linux pikvm 5.4.61-2-ARCH #1 SMP PREEMPT Fri Sep 4 12:46:16 UTC 2020 armv7l GNU/Linux
  
**journalctl -u kvm when I connect the iso:**

Dec 24 21:09:33 pikvm kvmd[369]: kvmd.plugins.msd.otg.helpers      INFO --- Unlocking the drive ...
Dec 24 21:09:33 pikvm kvmd[369]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-helper-otgmsd-unlock', 'unlock'] ...
Dec 24 21:09:33 pikvm sudo[3499]: pam_systemd_home(sudo:account): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Dec 24 21:09:33 pikvm sudo[3499]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-unlock unlock
Dec 24 21:09:33 pikvm sudo[3499]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Dec 24 21:09:33 pikvm sudo[3499]: pam_unix(sudo:session): session closed for user root
Dec 24 21:09:33 pikvm kvmd[369]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Sending SIGUSR1 to MSD 'file-storage' kernel thread with pid=367 ...

**journalctl -u kvm when I disconnect the iso:**
Dec 24 21:09:37 pikvm kvmd[369]: kvmd.plugins.msd.otg.helpers      INFO --- Unlocking the drive ...
Dec 24 21:09:37 pikvm kvmd[369]: kvmd.plugins.msd.otg.helpers      INFO --- Executing helper ['/usr/bin/sudo', '--non-interactive', '/usr/bin/kvmd-helper-otgmsd-unlock', 'unlock'] ...
Dec 24 21:09:37 pikvm sudo[3508]: pam_systemd_home(sudo:account): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Dec 24 21:09:37 pikvm sudo[3508]:     kvmd : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/kvmd-helper-otgmsd-unlock unlock
Dec 24 21:09:37 pikvm sudo[3508]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Dec 24 21:09:38 pikvm sudo[3508]: pam_unix(sudo:session): session closed for user root
Dec 24 21:09:38 pikvm kvmd[369]: kvmd.plugins.msd.otg.helpers      INFO --- Console: Sending SIGUSR1 to MSD 'file-storage' kernel thread with pid=367 ...

mdevaev, 2020-12-25:
Perhaps your ISO is larger than the supported size of 2.2 gigabytes?

valeech, 2020-12-25:
I was following the Tips section here: https://github.com/pikvm/pikvm/blob/master/pages/cookbook.md#create-a-microsoft-windows-based-flash-disk-image

Are bootable iso images limited to 2.2G? I thought I had read somewhere that if you switched from cdrom mode to flash mode you could use larger images. I am trying to create a Win10 bootable iso and my current iso is about 4.9GB. I created that from the Media Creation Tool...

Thanks for your help!

mdevaev, 2020-12-26:
I believe that flash drive mode and normal windows images may not be compatible. Alas, I don't know much about windows. If you have any more questions feel free to reopen this issue (or create a new one).

closed 2020-12-26 by mdevaev