pikvm/pikvm #140: @ key not working properly

PiTronica, 2020-12-26:
**Describe the bug**
When typing the @ in the stream i get ^@ not @ i need it without the ^

**To Reproduce**
connected to a 2nd raspberry pi with an hdmi video usb capture thingy
using a usb c splitter board
keyboard layout: belgian AZERTY

**Expected behavior**
typing @ 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: W10
 - Browser: FF
 - Version: 84

**Pi-KVM info:**
  - Raspberry Pi board version RPi 4
  - Pi-KVM platform v2-hdmi
  - Video capture type  =USB bridge
  - KVMD version: `vmd 2.8-1
                              kvmd-platform-v2-hdmiusb-rpi4 2.8-1
                              kvmd-webterm 0.35-1`
  - uStreamer version: ustreamer 2.2-1
  - Linux kernel: `Linux pikvm 5.4.83-3-ARCH #1 SMP PREEMPT Wed Dec 23 19:06:31 UTC 2020 armv7l GNU/Linux`

**Additional context**
Add any other context about the problem here.
[image]

ComputerDoktor, 2020-12-30:
the general issue is that AltGr combinations do not work correctly with remote Raspi OS.
AltGr combinations are used in many non-US keyboard layouts, german in my case, UK etc.
see http://kbdlayout.info/ http://kbdlayout.info/features/flags
https://docs.microsoft.com/en-us/globalization/keyboards/kbduk.html
instructions to reproduce the problem in an US-en environment:
- change the remote keyboard layout to uk
- install regional keyboard layout UK and switch to that layout.
- type <AltGr>+<#> on the on-screen-keyboard osk.exe
- expected result: solidus/backslash.

ComputerDoktor, 2020-12-30:
workaround: select US-layout for both local and remote keyboard

mdevaev, 2021-01-07:
@PiTronica is that workaround working?

mdevaev, 2021-01-09:
Workaround confirmed

closed 2021-01-09 by mdevaev