pikvm/pikvm #141: v0, will an Leonardo Pro Micro ATmega32U4 8MHz 3.3V work? (without needing a logic level converter)

francescoemme, 2020-12-30:
Hi
will a ATmega32U4 3.3V Leonardo clone work?
Something like
https://www.ebay.it/itm/Leonardo-Pro-Micro-ATmega32U4-8MHz-3-3V-Replace-ATmega328-Arduino-Pro-Mini/292901042757?hash=item4432433a45:g:qSIAAOSwBPRcLeQQ

The idea will be to simplify the wiring and connect it straight to the Raspberry GPIO.
This should be supported by the Arduino IDE but I'm not sure if your bin file require a 16Mhz clock or can work also at 8Mhz.
Thanks

Best Regards
Francesco

mdevaev, 2021-01-01:
We provide the firmware source code so that you can modify it for your needs. Yes, it will work. You need to edit the [platformio.ini](https://github.com/pikvm/kvmd/tree/master/hid) project file to set the desired core frequency (`board_build.f_cpu = 8000000L` like [here](https://docs.platformio.org/en/latest/boards/atmelavr/pro8MHzatmega328.html)).

mdevaev, 2021-01-08:
Feel free to reopen if you need it

closed 2021-01-08 by mdevaev