pikvm/pikvm #143: Does gpiozero work in pikvm?

libram11, 2021-01-01:
I have python installed but none of the commands I'm aware of work to install gpiozero. I'm trying to add a simple safe shutdown button since im getting corruption on the fat portion of the sd card. Am I just missing something or is it not supported?

mdevaev, 2021-01-01:
I don't understand, you can't install a library or so?

libram11, 2021-01-01:
yes cant get gpiozero to even run, i dont have to use that library if there is already any support in the build for a shutdown button though. I tried apt-get, pacman, direct library download, none worked at all. It could also be my inexperience with arch linux. So if it should work thats all I need to know to keep trying :)

mdevaev, 2021-01-01:
Gpiozero uses the deprecated kernel interface to gpio. I recommend you to use [libgpiod](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/). Pi-KVM uses it for own purposes (ATX and other things). libgpiod has official python binding.

libram11, 2021-01-01:
awesome ty

closed 2021-01-01 by mdevaev