pikvm/pikvm #144: Build breaks if password/key with trailing $ used

mnotgninnep, 2021-01-01:
Somewhere between make and docker something is getting mangled if you try to use a password or network key with a trailing $ symbol. Let's say my password is Password1$, it drops the $ from the end and the trailing ' resulting in either errors or weird behavious. I've tried escaping it and quoting in vaious ways, none of which have allowed the build to proceed. I should mention that passwords without a $ work and build fine.

I've tried the following combinations in config.mk:
WIFI_PASSWD = Password1$
Password1$$
Password1$$$$
Password1\$
Password1\$\$
All the above surrounded by " "
All the above surrounded by ' '
All the above surrounded by \" \"
All the above surrounded by \' \'

Results range from the sublime to the ridiculous:

Here is the output of 3 of the examples in increasing weirdness. I won't post all as they didn't vary much beyond this. Any way around there doesn't seem to be a way of escaping the $ to allow it to build this wifi network password.

BOARD = rpi3
PLATFORM = v0-hdmi
HOSTNAME = pikvm
LOCALE = en_GB
TIMEZONE = Europe/London
ROOT_PASSWD = Password1$
WEBUI_ADMIN_PASSWD = Password1$
IPMI_ADMIN_PASSWD = Password1$
WIFI_ESSID = mpennington
WIFI_PASSWD = Password1$

Gives the following ouput for make and docker. Note the missing $ and ' in the docker command line.

make -C ./.pi-builder os \
	NC= \
	BUILD_OPTS="  \
		--build-arg PLATFORM=v0-hdmi \
		--build-arg USTREAMER_VERSION=2.2-1 \
		--build-arg KVMD_VERSION=2.10-1 \
		--build-arg KVMD_WEBTERM_VERSION=0.35-1 \
		--build-arg WIFI_ESSID='mywifi' \
		--build-arg WIFI_PASSWD='Password1$' \
		--build-arg WIFI_IFACE='wlan0' \
		--build-arg ROOT_PASSWD='Password1$' \
		--build-arg WEBUI_ADMIN_PASSWD='Password1$' \
		--build-arg IPMI_ADMIN_PASSWD='Password1$' \
		--build-arg NEW_HTTPS_CERT=d582024c-38a9-4569-8c20-1557ba7e01c1 \
	" \
	PROJECT=pikvm-os-v0-hdmi \
	BOARD=rpi3 \
	STAGES='__init__ os pikvm-repo watchdog ro no-audit pikvm pikvm-image __cleanup__' \
	HOSTNAME=pikvm \
	LOCALE=en_GB \
	TIMEZONE=Europe/London \
	REPO_URL=http://de3.mirror.archlinuxarm.org

docker build \
		--rm \
		--tag pikvm-os-v0-hdmi-pi-builder-arm-result-rpi3 \
		 \
		 \
		--build-arg "BOARD=rpi3" \
		--build-arg "ARCH=arm" \
		--build-arg "LOCALE=en_GB" \
		--build-arg "TIMEZONE=Europe/London" \
		--build-arg "REPO_URL=http://de3.mirror.archlinuxarm.org" \
		--build-arg "PIKVM_REPO_URL=https://pikvm.org/repos" \
		--build-arg "PIKVM_REPO_KEY=912C773ABBD1B584" \
		--build-arg "REBUILD=065cccc3-4972-4e20-bae1-f39e3489dcf2" \
		--build-arg PLATFORM=v0-hdmi 		--build-arg USTREAMER_VERSION=2.2-1 		--build-arg KVMD_VERSION=2.10-1 		--build-arg KVMD_WEBTERM_VERSION=0.35-1 		--build-arg WIFI_ESSID='mywifi' 		--build-arg WIFI_PASSWD='Password1 		--build-arg WIFI_IFACE='wlan0' 		--build-arg ROOT_PASSWD='Password1 		--build-arg WEBUI_ADMIN_PASSWD='Password1 		--build-arg IPMI_ADMIN_PASSWD='Password1 		--build-arg NEW_HTTPS_CERT=d582024c-38a9-4569-8c20-1557ba7e01c1 	 \
	./.build

#############
[... 85 lines cut ...]
                PROJECT=pikvm-os-v2-hdmi \
                BOARD=zerow \
                STAGES='__init__ os pikvm-repo watchdog ro no-audit pikvm pikvm-image __cleanup__' \
                HOSTNAME=pikvm \
                LOCALE=en_GB \
                TIMEZONE=Europe/London \
                REPO_URL=http://de3.mirror.archlinuxarm.org

Which in turn gives

docker build \
                                --rm \
                                --tag pikvm-os-v2-hdmi-pi-builder-arm-result-zerow \
                                \
                                \
                                --build-arg "BOARD=zerow" \
                                --build-arg "ARCH=arm" \
                                --build-arg "LOCALE=en_GB" \
                                --build-arg "TIMEZONE=Europe/London" \
                                --build-arg "REPO_URL=http://de3.mirror.archlinuxarm.org" \
                                --build-arg "PIKVM_REPO_URL=https://pikvm.org/repos" \
                                --build-arg "PIKVM_REPO_KEY=912C773ABBD1B584" \
                                --build-arg "REBUILD=0748ef75-e2de-42f3-ae01-0f81277a5787" \
                                --build-arg PLATFORM=v2-hdmi                 --build-arg USTREAMER_VERSION=2.2-1                                --build-arg KVMD_VERSION=2.9-1                               --build-arg KVMD_WEBTERM_VERSION=0.35-1                   --build-arg WIFI_ESSID='mywifi'                         --build-arg WIFI_PASSWD='Password131897'                       --build-arg WIFI_IFACE='wlan0'                        --build-arg ROOT_PASSWD='Password131897'                    --build-arg WEBUI_ADMIN_PASSWD='Password131897' --build-arg IPMI_ADMIN_PASSWD='Password131897'                            --build-arg NEW_HTTPS_CERT=241c2e08-c6e4-4444-a847-05180cd36ffe    \
                ./.build

mdevaev, 2021-01-02:
With [this](https://github.com/pikvm/os/pull/10) patch it can work properly. Use double quotes for your password like `WIFI_PASSWD="test test test"`.

Remove `os` dir and checkout it again.

mnotgninnep, 2021-01-02:
Thanks for taking a look. It's still not working right.

BOARD = zerow
PLATFORM = v2-hdmi
HOSTNAME = pikvm
LOCALE = en_GB
TIMEZONE = Europe/London
ROOT_PASSWD = "Password1$"
WEBUI_ADMIN_PASSWD = "Password1$"
IPMI_ADMIN_PASSWD = "Password1$"
WIFI_ESSID = mywifi
WIFI_PASSWD = "Password1$"

make -C ./.pi-builder os \
	NC= \
	BUILD_OPTS='  \
		--build-arg PLATFORM=v2-hdmi \
		--build-arg USTREAMER_VERSION=2.2-1 \
		--build-arg KVMD_VERSION=2.10-1 \
		--build-arg KVMD_WEBTERM_VERSION=0.35-1 \
		--build-arg WIFI_ESSID=mywifi \
		--build-arg WIFI_PASSWD="Password1 \
		--build-arg WIFI_IFACE=wlan0 \
		--build-arg ROOT_PASSWD="Password1 \
		--build-arg WEBUI_ADMIN_PASSWD="Password1 \
		--build-arg IPMI_ADMIN_PASSWD="Password1 \
		--build-arg NEW_HTTPS_CERT=4a9f83b7-e403-466b-a847-cd9370490429 \
	' \
	PROJECT=pikvm-os-v2-hdmi \
	BOARD=zerow \
	STAGES='__init__ os pikvm-repo watchdog ro no-audit pikvm pikvm-image __cleanup__' \
	HOSTNAME=pikvm \
	LOCALE=en_GB \
	TIMEZONE=Europe/London \
	REPO_URL=http://de3.mirror.archlinuxarm.org

docker build \
		--rm \
		--tag pikvm-os-v2-hdmi-pi-builder-arm-result-zerow \
		 \
		 \
		--build-arg "BOARD=zerow" \
		--build-arg "ARCH=arm" \
		--build-arg "LOCALE=en_GB" \
		--build-arg "TIMEZONE=Europe/London" \
		--build-arg "REPO_URL=http://de3.mirror.archlinuxarm.org" \
		--build-arg "PIKVM_REPO_URL=https://pikvm.org/repos" \
		--build-arg "PIKVM_REPO_KEY=912C773ABBD1B584" \
		--build-arg "REBUILD=f061d221-b454-4371-947c-b9d567a80df5" \
		\
	--build-arg PLATFORM=v2-hdmi \
	--build-arg USTREAMER_VERSION=2.2-1 \
	--build-arg KVMD_VERSION=2.10-1 \
	--build-arg KVMD_WEBTERM_VERSION=0.35-1 \
	--build-arg WIFI_ESSID=mywifi \
	--build-arg WIFI_PASSWD="Password1 \
	--build-arg WIFI_IFACE=wlan0 \
	--build-arg ROOT_PASSWD="Password1 \
	--build-arg WEBUI_ADMIN_PASSWD="Password1 \
	--build-arg IPMI_ADMIN_PASSWD="Password1 \
	--build-arg NEW_HTTPS_CERT=4a9f83b7-e403-466b-a847-cd9370490429 \
 \
	./.build

mdevaev, 2021-01-03:
Don't run `make -C ./.pi-builder os`. Use `make os` in the `os` directory.

mnotgninnep, 2021-01-03:
That was make os. I was just trying to give you relevant snippets. Here's the full output:

make os
rm -rf ./.pi-builder/stages/{pikvm,pikvm-image,pikvm-otg-console}
cp -a pikvm pikvm-image pikvm-otg-console ./.pi-builder/stages
make -C ./.pi-builder os \
	NC= \
	BUILD_OPTS='  \
		--build-arg PLATFORM=v2-hdmi \
		--build-arg USTREAMER_VERSION=2.2-1 \
		--build-arg KVMD_VERSION=2.10-1 \
		--build-arg KVMD_WEBTERM_VERSION=0.35-1 \
		--build-arg WIFI_ESSID=mywifi \
		--build-arg WIFI_PASSWD="Password1 \
		--build-arg WIFI_IFACE=wlan0 \
		--build-arg ROOT_PASSWD="Password1 \
		--build-arg WEBUI_ADMIN_PASSWD="Password1 \
		--build-arg IPMI_ADMIN_PASSWD="Password1 \
		--build-arg NEW_HTTPS_CERT=fb2cbeaa-a924-450d-b534-09a5207965de \
	' \
	PROJECT=pikvm-os-v2-hdmi \
	BOARD=zerow \
	STAGES='__init__ os pikvm-repo watchdog ro no-audit pikvm pikvm-image __cleanup__' \
	HOSTNAME=pikvm \
	LOCALE=en_GB \
	TIMEZONE=Europe/London \
	REPO_URL=http://de3.mirror.archlinuxarm.org
make[1]: Entering directory '/home/michael/Desktop/pikvm/zerow/os/.pi-builder'
===== Ensuring toolbox image =====
docker build \
		--rm \
		--tag pi-builder-arm-toolbox \
		 \
		--file toolbox/Dockerfile.root \
	toolbox
Sending build context to Docker daemon  15.36kB
Step 1/5 : FROM archlinux/base
 ---> a79c9d41e4ac
Step 2/5 : RUN pacman --noconfirm -Syu 	&& pacman --needed --noconfirm -S 		base-devel 	arp-scan 		python 		parted 		dosfstools 		rsync 	&& (pacman --noconfirm -Sc || true)
 ---> Using cache
 ---> c2168eb235e7
Step 3/5 : RUN mkdir /tools
 ---> Using cache
 ---> 12815924a493
Step 4/5 : COPY install-binfmt /tools/
 ---> Using cache
 ---> c56d8e47e4d3
Step 5/5 : COPY docker-extract /tools/
 ---> Using cache
 ---> 98bebff3605a
Successfully built 98bebff3605a
Successfully tagged pi-builder-arm-toolbox:latest
===== Toolbox image is ready =====
===== Ensuring arm binfmt =====
docker run \
		--rm \
		--tty \
		--privileged \
	pi-builder-arm-toolbox /tools/install-binfmt \
		--mount \
		arm \
		/usr/bin/qemu-arm-static
:: Mounting binfmt_misc to '/proc/sys/fs/binfmt_misc' ...
:: Checking ARM binfmt configuration ...
:: Found existent ARM binfmt handler
:: ARM binfmt handler '/usr/bin/qemu-arm-static' is ready
===== Binfmt arm is ready =====
===== Ensuring base rootfs =====
if [ ! -e ./.tmp/base-rootfs-zerow.tar.gz ]; then \
	mkdir -p ./.tmp \
[... 242 lines cut ...]
 ---> a46feeae1c19
Step 66/87 : ARG WIFI_IFACE
 ---> Using cache
 ---> d7bfbc114ca1
Step 67/87 : RUN [ -z "$WIFI_ESSID" ] || ( 		export config="/etc/netctl/$WIFI_IFACE-${WIFI_ESSID/ /_}" 		&& echo "Description='Generated by Pi-KVM OS build system'" > $config 		&& echo "Interface='$WIFI_IFACE'" >> $config 		&& echo "Connection=wireless" >> $config 	&& echo "Security=wpa" >> $config 		&& echo "ESSID='$WIFI_ESSID'" >> $config 		&& echo "IP=dhcp" >> $config 		&& echo "Key='$WIFI_PASSWD'" >> $config 		&& systemctl enable netctl-auto@$WIFI_IFACE.service 	)
 ---> Using cache
 ---> 773be3edc309
Step 68/87 : ARG ROOT_PASSWD
 ---> Using cache
 ---> 49cd100f7550
Step 69/87 : ENV ROOT_PASSWD $ROOT_PASSWD
 ---> Using cache
 ---> 1210b747932a
Step 70/87 : RUN echo "root:$ROOT_PASSWD" | chpasswd 	&& echo "PermitRootLogin yes" >> /etc/ssh/sshd_config 	&& userdel -r -f alarm
 ---> Running in 368084a18bb9
No password has been supplied.
No password has been supplied.
No password has been supplied.
chpasswd: (user root) pam_chauthtok() failed, error:
Authentication token manipulation error
chpasswd: (line 1, user root) password not changed
The command '/bin/sh -c echo "root:$ROOT_PASSWD" | chpasswd 	&& echo "PermitRootLogin yes" >> /etc/ssh/sshd_config 	&& userdel -r -f alarm' returned a non-zero code: 1
make[1]: *** [Makefile:231: os] Error 1
make[1]: Leaving directory '/home/michael/Desktop/pikvm/zerow/os/.pi-builder'
make: *** [Makefile:54: os] Error 2

mdevaev, 2021-01-07:
I'll check it today. Sorry for the long reply.

mdevaev, 2021-01-12:
In the latest os/pi-builder: `WIFI_PASSWD = Password1$$$$` for `Password1$`. Remove old `os` environment, clone it again and retry.

mnotgninnep, 2021-01-13:
Thank you. I think that is working now.

closed 2021-01-13 by mnotgninnep