pikvm/pikvm #146: Unable to access bios config menu during boot

JuanEsc13, 2021-01-08:
**Describe the bug**
Unable to access bios config menu during boot. F keys seem to not work during boot neither on virtual keyboard or actual keyboard.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Reboot the server
2. Wait until bios boot screen appears
3. Try pressing any of the F keys on either virtual or real keyboard.

**Expected behavior**
When pressing F2 or F11 on either virtual or real keyboard during bios boot I should be taken to the Bios/boot menu.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 latest build
 - Browser Chrome
 - Version 87.0.4280.88

**Pi-KVM info:**
  - Raspberry Pi board versio:  RPi 4 8 GB
  - Pi-KVM platform: v2-hdmi
  - Video capture type: USB HDMI capture card
  - KVMD version: 
              kvmd 2.8-1
              kvmd-platform-v2-hdmiusb-rpi4 2.8-1
              kvmd-webterm 0.35-1
  - uStreamer version: ustreamer 2.2-1
  - Linux kernel: Linux pikvm 5.4.83-1-ARCH #1 SMP PREEMPT Mon Dec 14 15:43:54 UTC 2020 armv7l GNU/Linux

mdevaev, 2021-01-08:
I have heard that sometimes the DELL BIOS works strangely because it does not properly implement USB Composite. I have two possible tips:
* Try to update BIOS.
* If you can, try [Arduino HID](https://github.com/pikvm/pikvm/blob/master/pages/arduino_hid.md).

JuanEsc13, 2021-01-08:
You gave me the idea to plug the USB cable on one of the built in USB ports of the R710 server instead of the USB hub connected via PCIe and it is working now. Thank you!

closed 2021-01-08 by JuanEsc13

mdevaev, 2021-01-08:
Glad to hear it! :)

krasin, 2021-01-08:
@JuanEsc13 I am just a bystander here, but would like to say thank you for documenting your workaround here. It's a great public service that we all appreciate.

mdevaev, 2021-01-08:
I also documented it in the troubleshooting guide.

yoGhurrt1, 2021-01-12:
You have mentioned Dell BIOS, but how about HP? I have exact same problem with my N54L Proliant - tried on two different PCs with Mozilla 84. Raspberry is connected through https://www.tindie.com/products/8086net/usb-cpwr-splitter/ to server and charger, so it couldn't be USB hub. What's more interesting, it worked for a moment, after few days I started to have stability problems and maybe it was after some or other update.

mdevaev, 2021-01-13:
I don't think this is related to the update as these subsystems are almost unchanged. Glitches do not depend on the manufacturer, sometimes just bad luck.

yoGhurrt1, 2021-01-13:
To be clear, I meant Pi-KVM updates, not BIOS updates. Would Debbug log show anything related to this problem?

mdevaev, 2021-01-13:
See `journalctl -u kvmd`

yoGhurrt1, 2021-01-13:
Thanks. Here is my log from a moment when I switched resolution after rebooting Pi-KVM up to moment when I closed window because server didn't respond for F key and system boot started.

> Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.588    stream] -- Device fd=8 opened
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.589    stream] -- Using input channel: 0
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.590    stream] -- Using TV standard: DEFAULT
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.591      http] -- Starting HTTP eventloop ...
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.593    stream] -- Using resolution: 1280x1024
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.594    stream] -- Using pixelformat: JPEG
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.596    stream] -- Using HW FPS: 30
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.596    stream] -- Using IO method: MMAP
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.606    stream] -- Requested 5 device buffers, got 5
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.612    stream] -- Capturing started
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.613    stream] -- Switching to HW encoder because the input format is (M)JPEG
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [34.614    stream] -- Device does not support setting of HW encoding quality parameters
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.614    stream] -- Using JPEG quality: encoder default
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.615    stream] -- Creating pool with 1 workers ...
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.616    stream] -- Capturing ...
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Got SIGUSR2, checking the stream state ...
Jan 13 19:56:47 pikvm kvmd[362]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.114/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0'
Jan 13 19:56:47 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [34.652      http] -- HTTP: Registered client: [localhost]:0, id=bee6b60f-ef3c-4e9a-b40b-39dadfd892a1; clients now: 1
Jan 13 19:56:57 pikvm systemd[1]: systemd-hostnamed.service: Succeeded.
Jan 13 19:58:07 pikvm kernel: configfs-gadget gadget: error in submission: ep3out --> -11
Jan 13 19:58:07 pikvm kernel: dwc2 fe980000.usb: new device is high-speed
Jan 13 19:58:07 pikvm kernel: dwc2 fe980000.usb: new address 2
Jan 13 19:58:07 pikvm kernel: configfs-gadget gadget: high-speed config #1: c
Jan 13 19:58:48 pikvm kernel: configfs-gadget gadget: End Point Request ERROR: -108
Jan 13 19:58:48 pikvm kernel: configfs-gadget gadget: End Point Request ERROR: -108
Jan 13 19:58:48 pikvm kernel: configfs-gadget gadget: error in submission: ep3out --> -11
Jan 13 19:58:48 pikvm kernel: dwc2 fe980000.usb: new device is high-speed
Jan 13 19:58:48 pikvm kvmd[362]: kvmd.apps.kvmd.server             INFO --- Removed client socket: WsClient(id=3011098968, stream=True); clients now: 0
Jan 13 19:58:48 pikvm kvmd[362]: aiohttp.access                    INFO --- [admin (token) / 192.168.1.118] 'GET /ws HTTP/1.0' => 101; size=0 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0'
Jan 13 19:58:48 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Planning to stop streamer in 10.00 seconds ...
Jan 13 19:58:48 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [156.130      http] -- HTTP: Disconnected client: [localhost]:0, id=bee6b60f-ef3c-4e9a-b40b-39dadfd892a1, Success (reading,eof); clients now: 0
Jan 13 19:58:48 pikvm kvmd[362]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.114/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0'
Jan 13 19:58:48 pikvm kvmd[362]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=193 --- referer='https://192.168.1.114/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0'
Jan 13 19:58:48 pikvm kernel: dwc2 fe980000.usb: new address 1
Jan 13 19:58:49 pikvm kernel: configfs-gadget gadget: high-speed config #1: c
Jan 13 19:58:49 pikvm kvmd[362]: aiohttp.access                    INFO --- [admin (token) / 192.168.1.118] 'GET /info?fields=meta,extras HTTP/1.0' => 200; size=1490 --- referer='https://192.168.1.114/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0'
[... cut ...]
Jan 13 19:58:55 pikvm su[542]: (to root) kvmd-webterm on pts/0
Jan 13 19:58:55 pikvm su[542]: pam_unix(su-l:session): session opened for user root(uid=0) by (uid=969)
Jan 13 19:58:56 pikvm kernel: EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Jan 13 19:58:58 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Stopping streamer after delay ...
Jan 13 19:58:59 pikvm kvmd[362]: kvmd.apps.kvmd.streamer           INFO --- Streamer killed: pid=442; retcode=0

Beside "configfs-gadget gadget: error in submission: ep3out --> -11" nothing looks suspicious.

mdevaev, 2021-01-17:
If you have another memory card, install the old OS image from the site there and try it. So we can understand whether the update is to blame.

yoGhurrt1, 2021-01-21:
That's a good point. I'll search for older images and go back to you.