pikvm/pikvm #147: tailscale requires readwrite filesystem

moodoki, 2021-01-08:
**Describe the bug**
tailscale seems to keep an open handle on some files.
Also, tailscale will not start if filesystem is readonly (i.e. after reboot)

**To Reproduce**
Steps to reproduce the behavior, like:
1. Install and setup tailscale as per [documentation](https://github.com/pikvm/pikvm/blob/master/pages/tailscale.md)
2. After the steps, `ro` command will fail. Showing `mount: /: mount point is busy.`
3. After `reboot`, tailscale will fail to start with these messages in `journalctl`
```
Sep 04 15:45:30 pikvm systemd[1]: Starting Tailscale node agent...
Sep 04 15:45:31 pikvm tailscaled[330]: 2020/09/04 15:45:31 --state is required
Sep 04 15:45:31 pikvm systemd[1]: tailscaled.service: Control process exited, code=exited, status=1/FAILURE
Sep 04 15:45:31 pikvm tailscaled[347]: 2020/09/04 15:45:31 --state is required
Sep 04 15:45:31 pikvm systemd[1]: tailscaled.service: Control process exited, code=exited, status=1/FAILURE
Sep 04 15:45:31 pikvm systemd[1]: tailscaled.service: Failed with result 'exit-code'.
Sep 04 15:45:31 pikvm systemd[1]: Failed to start Tailscale node agent.
Sep 04 15:45:31 pikvm systemd[1]: tailscaled.service: Scheduled restart job, restart counter is at 1.
Sep 04 15:45:31 pikvm systemd[1]: Stopped Tailscale node agent.
Sep 04 15:45:31 pikvm systemd[1]: Starting Tailscale node agent...
Sep 04 15:45:31 pikvm tailscaled[353]: 2020/09/04 15:45:31 --state is required
Sep 04 15:45:31 pikvm systemd[1]: tailscaled.service: Control process exited, code=exited, status=1/FAILURE
Sep 04 15:45:31 pikvm tailscaled[360]: 2020/09/04 15:45:31 --state is required
Sep 04 15:45:31 pikvm systemd[1]: tailscaled.service: Control process exited, code=exited, status=1/FAILURE
Sep 04 15:45:31 pikvm systemd[1]: tailscaled.service: Failed with result 'exit-code'.
...
```
`tailscaled` service would start successfully if file system is set to rw

**Expected behavior**
`ro` command should succeed.
`tailscaled` should start after reboot
`rw` should not be required to start `tailscaled`

**Pi-KVM info:**
  - Raspberry Pi board version - RPi 4
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: `kvmd 1.100-1
kvmd-platform-v2-hdmi-rpi4 1.100-1
kvmd-webterm 0.34-1
`
  - uStreamer version: `ustreamer 1.25-1`
  - Linux kernel: Linux pikvm 5.4.61-2-ARCH #1 SMP PREEMPT Fri Sep 4 12:46:16 UTC 2020 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

mdevaev, 2021-01-08:
Update the whole system using `rw; pacman -Syu; reboot` and try again.

moodoki, 2021-01-08:
Updating the whole system doesn't help. The error still remains.

moodoki, 2021-01-08:
It looks like `tailscaled` has open handles on `/var/lib/tailscale/tailscaled.log1.txt` and `/var/lib/tailscale/tailscaled.log2.txt`.
I don't see a way to change the path of these files. Any ideas?

mdevaev, 2021-01-08:
I don't really understand where this problem came from at all because I haven't updated the package in the last couple of months, and all this time it worked fine. I'll figure it out, just today I had a higher priority issue in the form of a broken csi bridge driver.

moodoki, 2021-01-09:
@mdevaev No worries, I'll try to debug as well. 

Looks like from a recent version of tailscale, they've decided that it was a good idea to write to 2 files in `/var/lib/tailscale`. https://github.com/tailscale/tailscale/blob/f9659323df677a35dc1c53bcb6c55d708933571b/logpolicy/logpolicy.go#L111

I'm not sure why they've decided to do so instead of using LOGS_DIRECTORY. STATE_DIRECTORY doesn't seem to be used anywhere else in tailscale.

I've verified that I can get it to start using a writable overlay, with something like this:
```
#!/bin/bash

mkdir -p /var/log/tailscale.overlay/upper
mkdir -p /var/log/tailscale.overlay/workdir
mount -t overlay -o lowerdir=/var/lib/tailscale.ro,upperdir=/var/log/tailscale.overlay/upper,workdir=/var/log/tailscale.overlay/workdir overlay /var/lib/tailscale
```

mdevaev, 2021-01-09:
The problems are related to the fact that tailscale was accepted into the official arch repository. Systemd service is not patched there, as it was with me. As a workaround, do the following:
1. `mkdir -p /etc/systemd/system/tailscaled.service.d`
2. Create file `/etc/systemd/system/tailscaled.service.d/override.conf`:
    ```
    [Service]
    ExecStartPre=
    ExecStartPre=/usr/sbin/tailscaled --cleanup --state=/var/lib/tailscale/tailscaled.state
    ExecStopPost=
    ExecStopPost=/usr/sbin/tailscaled --cleanup --state=/var/lib/tailscale/tailscaled.state
    StateDirectory=
    StateDirectoryMode=
    ```

mdevaev, 2021-01-09:
Alternative: `pacman -S tailscale-pikvm`

mdevaev, 2021-01-09:
I've fixed doc. Problem solved.

closed 2021-01-09 by mdevaev

moodoki, 2021-01-09:
Awesome. Verified that it works! Thanks.