pikvm/pikvm #148: Direct download to Pi for mass storage

theLMGN, 2021-01-09:
**Is your feature request related to a problem? Please describe.**
It's a waste of time and annoying when I want to use the mass storage feature as I have to download the iso to my local machine, then back up to the Pi

**Describe the solution you'd like**
Add a direct download box, like a more polished version of the follwing 
[image]

**Describe alternatives you've considered**
 - Downloading to my local host then to up to Pi-KVM, too slow, requires storage space on the local machine
 - Downloading to the Pi-KVM using wget in terminal, not immediately obvious, not user friendly.

mdevaev, 2021-01-09:
This is a good idea, but I will consider its implementation a low priority, since there is now a documented way to load images directly to the Pi via the terminal: https://github.com/pikvm/pikvm/blob/master/pages/cookbook.md#upload-iso-images-manually

mdevaev, 2021-07-27:
Working on it now.

mdevaev, 2021-08-01:
Will be available in the next release, follow news.

closed 2021-08-01 by mdevaev