pikvm/pikvm #150: Intel vPro/AMT and AMD DASH as alternative to ATX wiring

kosmonavtika, 2021-01-12:
First, thanks for all your great work on this fine project, @mdevaev!

I'm interested in whether it would be possible to use the remote power management features of Intel vPro/AMT and AMD DASH as an alternative to ATX wiring and soldering. Simply put an easier to use and software based alternative for users who have Intel vPro/AMT or AMD DASH capable hardware. Lowing the barrier by allowing a simpler solution to power management may increase adoption?

My idea is that this could be implemented as a plugin in `kvmd`, similarly to how the ATX plugin is implemented now. 

Of course, this approach may present security issues, as I'm not really sure what best security practices are in relation to Intel vPro/AMT and AMD DASH. It may be that these features *should* be disabled for security reasons. It could also be the case that Intel vPro/AMT or AMD DASH doesn't implement all of the features currently available by ATX wiring, like `HDDLED` and `PLED`.

mdevaev, 2021-01-16:
Yes, it can be implemented like you described. But I don't have a single board to check.

JsBergbau, 2021-03-17:
Intel vPro/AMT works great to switch on PC. From my point of view: When you have Intel vPro/AMT you don't need PI KVM since you can use any VNC client so view monitor. Even switching between monitors is supported.

AMD Dash has no VNC implementation by default. I have a Lenovo ThinkCentre M70q Generation 2. No support for VNC. As soon as you enable DASH PC listens on Port 5900 / VNC. As soon as you connect to Port 5900 without any handshake, PC reboots, like by `nc 5900`. See full information there https://forums.lenovo.com/t5/ThinkCentre-A-E-M-S-Series/Some-information-about-AMD-Dash-Realtek-DASH-on-Lenovo-ThinkCentre-M75q/m-p/5069475

I had even the situation where I couldn't turn the PC via AMD Dash on. I tried it twice, but it didn't work.

So when you have AMD Dash you shouldn't enable it and solely use PI KVM. When you have Intel vPro you don't need it. So I would say save your time and invest it in other useful features.

mdevaev, 2022-04-30:
If these interfaces has CLI tools, they can be used through pseudo-gpio plugin [CMD](https://docs.pikvm.org/gpio/#cmd). I dont think they reqired pure ATX plugin right now.

closed 2022-04-30 by mdevaev

crahan, 2022-10-16:
Apologies for commenting on an old issue conversation, but I figured people who end up here after searching on how to configure this might want to know this can be set up per @mdevaev's suggestion, using https://pypi.org/project/amt/ (`openwsman` and `wsmancli` did not work unfortunately as these tools don't appear to be available as ARM packages in Arch Linux).

Below is an example screenshot of how I've set up group and individual on/off/reboot controls for 3 devices. AMT buttons are configured with confirmation dialogs as described at https://docs.pikvm.org/gpio/?h=wake#view to ensure an incorrect mouse click doesn't unintentionally take down the whole rack! 😊

[image]

mattague, 2024-03-12:
> Below is an example screenshot of how I've set up group and individual on/off/reboot controls for 3 devices. AMT buttons are configured with confirmation dialogs as described at [...]

Would you be able to show your override.yaml? I'm having trouble getting amt installed and working via the buttons in the gui.