pikvm/pikvm #151: Stream killed

UnivSim, 2021-01-13:
**Describe the bug**
A clear and concise description of what the bug is.
[image]

**To Reproduce**
Steps to reproduce the behavior, like:
Unknown

**Expected behavior**
Stream that works

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: W7
 - Browser FF
 - Version 84

**Pi-KVM info:**
  - Raspberry Pi board versionRPi 4
  - Pi-KVM platform  v2-hdmi
  - Video capture type USB
  - KVMD version: `kvmd 2.13-1`
  - uStreamer version: `ustreamer 3.1-1`
  - Linux kernel: `Linux pikvm 5.4.83-5-ARCH #1 SMP PREEMPT Sun Jan 3 16:11:27 UTC 2021 armv7l GNU/Linux`

**Additional context**
Add any other context about the problem here.

mdevaev, 2021-01-13:
Just perform system update and reboot. It's already fixed:
```
rw
pacman -Syu
reboot
```

ComputerDoktor, 2021-01-14:
I am getting this error too: Can't initialize OMX Core: OMX_ErrorHardware .... Streamer killed:
No luck with update: 
:: Starting full system upgrade...
resolving dependencies...
warning: cannot resolve "raspberrypi-firmware>=20210112-2", a dependency of "kvmd-platform-v2-hdmiusb-rpi4"
:: The following package cannot be upgraded due to unresolvable dependencies:
      kvmd-platform-v2-hdmiusb-rpi4

mdevaev, 2021-01-14:
```
pacman -S raspberrypi-firmware=20210112-2
pacman -Syu
reboot
```

PiTronica, 2021-01-14:
Unfortunatly this doesn't help me:

`[2021-01-14 22:30:51 kvmd.service] --- Traceback (most recent call last):
[2021-01-14 22:30:51 kvmd.service] ---   File "/usr/lib/python3.9/site-packages/kvmd/apps/kvmd/streamer.py", line 386, in __streamer_task_loop
[2021-01-14 22:30:51 kvmd.service] ---     raise RuntimeError("Streamer unexpectedly died")
[2021-01-14 22:30:51 kvmd.service] --- RuntimeError: Streamer unexpectedly died
[2021-01-14 22:30:52 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Streamer killed: pid=2119; retcode=-11
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Started streamer pid=2132: ['/usr/bin/ustreamer', '--device=/dev/kvmd-video', '--persistent', '--format=jpeg', '--resolution=1920x1080', '--desired-fps=30', '--drop-same-frames=30', '--last-as-blank=0', '--unix=/run/kvmd/ustreamer.sock', '--unix-rm', '--unix-mode=0660', '--exit-on-parent-death', '--process-name-prefix=kvmd/streamer', '--notify-parent', '--no-log-colors', "['--slowdown']", '--h264-sink=kvmd::ustreamer::h264']
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.958      main] -- Using internal blank placeholder
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.958      main] -- Using H264-sink: kvmd::ustreamer::h264
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.960      main] -- Listening HTTP on UNIX socket '/run/kvmd/ustreamer.sock'
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.960      http] -- Starting HTTP eventloop ...
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.961    stream] -- Using V4L2 device: /dev/kvmd-video
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.961    stream] -- Using desired FPS: 30
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.961    stream] -- H264: Initializing MMAL encoder ...
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.961    stream] -- H264: Using bitrate: 5000 Kbps
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.961    stream] -- H264: Using GOP: 30
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: mmalipc: mmal_vc_init_fd: could not open vchiq service
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: mmal: mmal_vc_component_create: failed to initialise mmal ipc for 'vc.ril.video_encode' (7:EIO)
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: mmal: mmal_component_create_core: could not create component 'vc.ril.video_encode' (7)
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [283.961    stream] -- H264: Can't create MMAL wrapper: MMAL_EIO [IO error]
[2021-01-14 22:30:53 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [283.962    stream] -- H264: Destroying MMAL encoder ...
[2021-01-14 22:30:54 kvmd.service] --- kvmd.apps.kvmd.streamer          ERROR --- Unexpected streamer error: pid=2132
[2021-01-14 22:30:54 kvmd.service] --- Traceback (most recent call last):
[2021-01-14 22:30:54 kvmd.service] ---   File "/usr/lib/python3.9/site-packages/kvmd/apps/kvmd/streamer.py", line 386, in __streamer_task_loop
[2021-01-14 22:30:54 kvmd.service] ---     raise RuntimeError("Streamer unexpectedly died")
[2021-01-14 22:30:54 kvmd.service] --- RuntimeError: Streamer unexpectedly died`

mdevaev, 2021-01-14:
You need to update to ustreamer 3.4. It has a fix for this

PiTronica, 2021-01-14:
This issue can be closed as it was fixed remotely.
(sorry for switching between accounts, seems i use a diferent one on my work PC)

mdevaev, 2021-01-16:
Bug fixed, update instructions are above.

closed 2021-01-16 by mdevaev