pikvm/pikvm #153: Strobing with EZCOO Switch

ibivibiv, 2021-01-16:
**Describe the bug**
On both a USB and Camera port build I get a strobing of the video when connecting an EZCOO KVM switch.  The EZCOO switch works fine with a TV and the PiKVM works fine direct connected.  The minute I connect the two no matter what type of device I connect to the EZCOO port the image over the webserver flashes.  I've tried a variety of resolution and refresh settings and it doesn't seem to make a difference.  I'm wondering if anyone else has had this happen?

**To Reproduce**
Steps to reproduce the behavior, like:
1. Connect devices to EZCOO KVM Switch
2. Connect Output of EZCOO to PiKVM
3. Start KVM in web console
4. Image Flashes from black to visible every few seconds then immediately back to black for a few seconds

**Expected behavior**
stable image

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Browser firefox
 - Version: pulled master
 - VNC client (if used)

**Pi-KVM info:**
  - Raspberry Pi board Pi 4
  - Pi-KVM platform = tried both
  - Video capture type tried both
  - KVMD version: 2.14-1
  - uStreamer version: 3.5.1
  - Linux kernel: Linux pikvm 5.4.83-5-ARCH #1 SMP PREEMPT Sun Jan 3 16:11:27 UTC 2021 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

ibivibiv, 2021-01-16:
On a whim I just put an hdmi audio normalizing box inline between the ezcoo and the HDMI in (USB right now).  It is working.  I am guessing something is afoul between the output of the ezcoo and the input of the PiKVM.  If you feel like closing this you can, at least this will help someone else in the future if they hit this same issue.  Maybe its a flaky ezcoo?  Something with that hotplugging mentioned in the markdown docs?

mdevaev, 2021-01-16:
I heard something from ezcoo users that if the switch is connected after loading pikvm, then everything will be fine. Can you confirm that?

mdevaev, 2021-01-28:
Sup?

mdevaev, 2021-02-16:
Hello?

mdevaev, 2021-02-17:
No response. Write if you want to continue debugging.

closed 2021-02-17 by mdevaev