pikvm/pikvm #156: cannot resolve "raspberrypi-firmware>=20210112-2"

kkellner, 2021-01-19:
When I try to update first time after burning image, I get the following.  Am I doing something wrong?

```
[root@pikvm ~]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
:: Replace libutil-linux with core/util-linux-libs? [Y/n] y
resolving dependencies...
warning: cannot resolve "raspberrypi-firmware>=20210112-2", a dependency of "kvmd-platform-v2-hdmi-rpi4"
:: The following package cannot be upgraded due to unresolvable dependencies:
      kvmd-platform-v2-hdmi-rpi4

:: Do you want to skip the above package for this upgrade? [y/N]
```

mdevaev, 2021-01-19:
The upstream distribution hosted a broken package, so I replaced it with my own. Just make `pacman -S raspberrypi-firmware=20210112-2` and run `pacman -Syu` again.

kkellner, 2021-01-20:
The update worked with the instructions above.   Thank you for your help.

closed 2021-01-20 by kkellner

mdevaev, 2021-01-20:
You're welcome

oswaldo-be, 2022-01-22:
Where am I the problem?
[root@pikvm ~]# rw
[root@pikvm ~]# pacman -S raspberrypi-firmware=20210112-2
error: target not found: raspberrypi-firmware=20210112-2

mdevaev, 2022-01-22:
This is not actual anymore

oswaldo-be, 2022-01-22:
But I do still have the from the thread starter mentioned problem.
So how can I resolve this?
please

mdevaev, 2022-01-22:
Please reflash the memory card. This is the easiest way.

oswaldo-be, 2022-01-22:
Oh thank you very much.
It would definitely be the easiest way, if the memory card was not 300km away

mdevaev, 2022-01-22:
Please run `pacman -Syu` and show me the log.

oswaldo-be, 2022-01-22:
warning: cannot resolve "python<3.10", a dependency of "kvmd"
warning: cannot resolve "python<3.10", a dependency of "kvmd"
warning: cannot resolve "kvmd=3.50-1", a dependency of "kvmd-platform-v2-hdmi-rpi4"
:: The following packages cannot be upgraded due to unresolvable dependencies:
      kvmd  kvmd-platform-v2-hdmi-rpi4

:: Do you want to skip the above packages for this upgrade? [y/N] y
looking for conflicting packages...
 there is nothing to do

tail...
[root@pikvm ~]# tail /etc/pacman.conf

# An example of a custom package repository.  See the pacman manpage for
# tips on creating your own repositories.
#[custom]
#SigLevel = Optional TrustAll
#Server = file:///home/custompkgs

[pikvm]
Server = https://pikvm.org/repos/rpi4-arm
SigLevel = Required DatabaseOptional

mdevaev, 2022-01-22:
Run `sed -i -e "s|^Server = https://pikvm.org/repos/|Server = https://files.pikvm.org/repos/arch/|g" /etc/pacman.conf` and retry `pacman -Syu` again.

oswaldo-be, 2022-01-22:
great, thank you very much. the error is gone

oswaldo-be, 2022-01-22:
now fails nginx with:

systemctl status nginx
x nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/usr/lib/systemd/system/nginx.service; disabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2022-01-22 18:49:32 MSK; 3s ago
    Process: 868 ExecStart=/usr/bin/nginx -g pid /run/nginx.pid; error_log stderr; (code=exited, status=1/FAILURE)
        CPU: 57ms

Jan 22 18:49:32 pikvm systemd[1]: Starting A high performance web server and a reverse proxy server...
Jan 22 18:49:32 pikvm nginx[868]: 2022/01/22 18:49:32 [warn] 868#868: could not build optimal types_hash, you sho>
Jan 22 18:49:32 pikvm nginx[868]: 2022/01/22 18:49:32 [emerg] 868#868: open() "/var/log/nginx/access.log" failed >
Jan 22 18:49:32 pikvm systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Jan 22 18:49:32 pikvm systemd[1]: nginx.service: Failed with result 'exit-code'.
Jan 22 18:49:32 pikvm systemd[1]: Failed to start A high performance web server and a reverse proxy server.

https://github.com/pikvm/pikvm/issues/280
did not help

mdevaev, 2022-01-22:
Don't run `nginx` service. PiKVM uses `kvmd-nginx` for web. Contact me via discord. I'll try to fix the error remotely for you.