pikvm/pikvm #157: HDMI passthrough ?

amplatzer, 2021-01-21:
Is it possible pikvm can also stream the video signal to the monitor?

mdevaev, 2021-01-21:
Hi! This is a feature that I often ask for. It can't be done right now, but I'm working on a solution to this problem. The case has moved forward and perhaps in the next couple of months you will get this.

wisteso, 2021-03-10:
how is the latency with the temporarily solution?

bozserz, 2021-03-10:
Complex issue. The device connected to HDMI1 output has a set of possible resolutions, but the stream is not necessarily the same. When I first connected the TV to the output, it was an approx. 640x480 stuff, but after a restart of the Pi, a much higher resolution was in effect. So maybe there will be a complicated description in the end, what to connect/start in which order to operate properly... Or is it possible to read the EDID data of the display connected later on the fly and modify the output settings accordingly?
The mplayer has to be installed first, as it's not included in the distro by default (it installs plenty of other packages). Also the scale (for the version I've installed) is requiring an integer only, so 1920x1080 gave an error for me (but worked with 1920 only).
The TV showed the screen contents in the upper-left corner... I will try it with a normal PC monitor (maybe that will be better, as the monitor will/can swich the resolution automatically, the TV does not).
The command above containing the mplayer line quits immediately, if a stream is not started (yet). Maybe it should auto-start upon a stream start/reset. Or have a button there, similar to the reset stream, to start/restart HDMI output (maybe monitoring the state), so no console access should be necessary. But it seems, that this solution is really-really close.

mdevaev, 2021-03-13:
It's strange that it works so slowly. I will try to make my own implementation of the player for full-screen terminal output

petrkr, 2021-05-30:
What about use same way how raspivid do?

mdevaev, 2021-05-30:
We have a different architecture, so it will not work.

ctyler, 2021-06-01:
Possible solution: There are HDMI-USB capture devices with a passthrough. Example: https://www.amazon.ca/Capture-USB2-0-Streaming-Device-Nintendo/dp/B08997KNSG/ or search for "HDMI USB capture loop"

Alternatively, you could add an HDMI splitter before the feed to the capture device (CSI or USB).

petrkr, 2021-06-01:
Those are workarounds. 

Meaning was if you can play accelerated stream directly from CSI to HDMI output on Rpi, it would be handy if ustreamer will support show CSI input directly on HDMI output as raspivid does.

I do not had time to check source codes of both projects, but in short, what is problem to copy same stream to HDMI output and provide it also to MJPEG?

as I mention before, it would be only for CSI input, because it seems to be working on HW level

mdevaev, 2022-05-25:
https://raspberrypi.stackexchange.com/questions/82260/how-can-i-detect-the-hdmi-status-of-the-pi

SemperFi89, 2023-01-25:
> Possible solution: There are HDMI-USB capture devices with a passthrough. Example: https://www.amazon.ca/Capture-USB2-0-Streaming-Device-Nintendo/dp/B08997KNSG/ or search for "HDMI USB capture loop"
> 
> Alternatively, you could add an HDMI splitter before the feed to the capture device (CSI or USB).

I have this device, but PiKVM doesn't see it.  What config options do I need to set/change to make this work?  I have the official PiKVM Hat from Kickstarter.

zakx, 2024-02-01:
I bought several PiKVM v4 Plus devices, thinking that this feature would already exist. Is there a realistic timeline for proper support (that not being a restreamed video capture), and would the hardware even support this use-case? 

I am afraid that the suggested workaround is just not an option in my case.

mdevaev, 2024-02-01:
Sorry, we have delayed the implementation of this feature, but we hope to release it this year, possibly in Q1-Q2

mdevaev, 2024-02-02:
However, we have some progress. Alpha is working on V4:
[image]

IsaacHayes1995, 2024-03-19:
@mdevaev this is great news. We will test this feature for an upcoming deployment.

mdevaev, 2024-04-07:
A big update. Now the image on the physical screen will be available simultaneously with the web stream. Some reconfiguration of the services is required, so check the manual to use it: https://docs.pikvm.org/pass

ryanwwest, 2024-04-13:
Will it eventually be possible to support higher resolutions? Right now I understand the maximum is 1920x1200, but it would be nice to eventually get 1440p, 4k, and an oddball - 1920x2160 (one half of a 4k display, which large monitors like LG 43UN700-B use to simultaneously output two different "2k" HDMI inputs to different halves of the screen at 60Hz even if 4k itself can only reach 30Hz).

mdevaev, 2024-04-13:
No, this is a hardware limitation

mdevaev, 2024-04-16:
Okay, finally it's released: https://www.reddit.com/r/pikvm/comments/1c5s2dh/kvmd_3333_it_will_not_pass_hdmi_passthrough/

closed 2024-04-16 by mdevaev