pikvm/pikvm #158: Ethernet wiznet w5500

pawkon, 2021-01-22:
Ethernet wiznet w5500 with SPI interface is not working. 
Module under Raspberry OS is working just fine but under pikvm it is not loading eth0 interface.
There is overlay for this module "w5500.dtbo" but loading it doesn't change anything. I tried like this:
dtoverlay=w5500
then like this:
dtparam=spi=on
dtoverlay=w5500
and then like this
dtoverlay=anyspi,spi0-0,dev="w5500",speed=30000000
dtoverlay=w5500
The second option is working just fine under RpiOS. 
I've noticed that in the pikvm os the w5100-spi module is missing but I don't know if it is responsible for activating this module - I have found that in some tutorials that it is being loaded
Please help me to resolve this issue.
I'm trying to make it run on Raspberry Pi Zero v 1.3
Thanks

bbeaudoin, 2021-01-23:
The `dtoverlay` lines specify device tree configuration for device discoverability. It's still necessary for the distribution maintainers to compile the modules and distribute the modules with the kernel.

It appears the module is already included at build time:
[https://github.com/archlinuxarm/PKGBUILDs/blob/bbe2e1cbd42515d1b9d6e5fc2a6ddc2956c2a16c/core/linux-armv7/config#L3150](https://github.com/archlinuxarm/PKGBUILDs/blob/bbe2e1cbd42515d1b9d6e5fc2a6ddc2956c2a16c/core/linux-armv7/config#L3150)

Checking the filesystem I see the w5100 module is available but I don't have the required hardare to confirm the report.

```bash
[root@pikvm lib]# find /lib/modules -name w51\*
/lib/modules/5.4.83-5-ARCH/build/include/config/wiznet/w5100.h
/lib/modules/5.4.83-5-ARCH/kernel/drivers/net/ethernet/wiznet/w5100.ko.gz
[root@pikvm lib]# 
```

Please consider opening a forum account for Arch Linux ARM upstream and asking about the issue. If you believe this is a bug with the Arch Linux ARM kernel distribution, you can submit a bug report directly on the [Arch Linux ARM Packages Forum](https://archlinuxarm.org/forum/viewforum.php?f=15).

pawkon, 2021-01-23:
I have posted a question on the arch linux forum. I hope someone will respond. I will keep posted here if something change if someone else will have the same problem. If anyone here have any idea what to do please give me a hint. Thanks

pawkon, 2021-01-24:
well ... currently pikvm uses 5.4.83-3 kernel and in 5.4.83-4 there is a patch to Arch Linux making w5500-spi module to compile. Probably this module is responsible to get the w5500 ethernet module to work. How to get the 5.4.83-4? what and where to change in pikvm compile options?

mdevaev, 2021-01-25:
5.4.83-7 anailable in the repo. Just update:
```
rw
pacman -Syu
reboot
```

pawkon, 2021-01-25:
Those commands should be issued under pikvm arch linux or in Ubuntu before make os command? If under pikvm then this is not possible because I'm running os on rpi zero v 1.3 which doesn't have internet connection. This is why I need SPI-Ethernet module to get working

mdevaev, 2021-01-25:
Okay, rebuild the image: `make os NC=1`.

pawkon, 2021-01-25:
with NC=1 I'm getting error:

`$ make os NC=1
git clone --depth=1 https://github.com/mdevaev/pi-builder ./.pi-builder
Cloning into './.pi-builder'...
remote: Enumerating objects: 50, done.
remote: Counting objects: 100% (50/50), done.
remote: Compressing objects: 100% (39/39), done.
remote: Total 50 (delta 0), reused 34 (delta 0), pack-reused 0
Unpacking objects: 100% (50/50), 3.98 MiB | 1.99 MiB/s, done.
rm -rf ./.pi-builder/stages/{pikvm,pikvm-image,pikvm-otg-console}
cp -a pikvm pikvm-image pikvm-otg-console ./.pi-builder/stages
make -C ./.pi-builder os \
	NC=1 \
	BUILD_OPTS='  \
		--build-arg PLATFORM=v2-hdmi \
		--build-arg USTREAMER_VERSION=3.13-1 \
		--build-arg KVMD_VERSION=2.15-1 \
		--build-arg KVMD_WEBTERM_VERSION=0.35-1 \
		--build-arg WIFI_ESSID= \
		--build-arg WIFI_PASSWD= \
		--build-arg WIFI_IFACE=wlan0 \
		--build-arg ROOT_PASSWD=root \
		--build-arg WEBUI_ADMIN_PASSWD=admin \
		--build-arg IPMI_ADMIN_PASSWD=admin \
		--build-arg NEW_HTTPS_CERT=18253a7e-dcad-4eb2-ac9d-d1d1c8ec5818 \
	' \
	PROJECT=pikvm-os-v2-hdmi \
	BOARD=zerow \
	STAGES='__init__ os pikvm-repo watchdog ro no-audit pikvm pikvm-image __cleanup__' \
	HOSTNAME=pikvm \
	LOCALE=en_US \
	TIMEZONE=Europe/Moscow \
	REPO_URL=http://de3.mirror.archlinuxarm.org
make[1]: Entering directory '/home/pilot/os/.pi-builder'
===== Ensuring toolbox image =====
docker build \
		--rm \
		--tag pi-builder-arm-toolbox \
		 \
		--file toolbox/Dockerfile.root \
	toolbox
Sending build context to Docker daemon  15.36kB
Step 1/6 : FROM archlinux/archlinux:base-devel
base-devel: Pulling from archlinux/archlinux
8c69f942fbe6: Pull complete 
146dd82a0154: Pull complete 
Digest: sha256:bcde265d6c1705f4483800720aa9e442b6f8a666421879674ee500fdeca8c9a4
Status: Downloaded newer image for archlinux/archlinux:base-devel
 ---> 5951ea871a44
Step 2/6 : RUN mkdir -p /etc/pacman.d/hooks 	&& ln -s /dev/null /etc/pacman.d/hooks/30-systemd-tmpfiles.hook
 ---> Running in 340c1ecda6ba
Removing intermediate container 340c1ecda6ba
 ---> 849b342197eb
Step 3/6 : RUN pacman --noconfirm -Syu 	&& pacman --needed --noconfirm -S 		arp-scan 		python 		parted 		dosfstools 		rsync 	&& (pacman --noconfirm -Sc || true) 	&& rm -rf /var/cache/pacman/pkg/*
 ---> Running in 73b4a1b03bd4
:: Synchronizing package databases...
error: failed retrieving file 'core.db' from mirror.pkgbuild.com : Resolving timed out after 10000 milliseconds
error: failed retrieving file 'core.db' from mirror.rackspace.com : Resolving timed out after 10000 milliseconds
error: failed retrieving file 'core.db' from mirror.leaseweb.net : Resolving timed out after 10001 milliseconds
error: failed to update core (download library error)
error: failed retrieving file 'extra.db' from mirror.pkgbuild.com : Resolving timed out after 10000 milliseconds
error: failed retrieving file 'extra.db' from mirror.rackspace.com : Resolving timed out after 10000 milliseconds
error: failed retrieving file 'extra.db' from mirror.leaseweb.net : Resolving timed out after 10000 milliseconds
error: failed to update extra (download library error)
error: failed retrieving file 'community.db' from mirror.pkgbuild.com : Resolving timed out after 10000 milliseconds
error: failed retrieving file 'community.db' from mirror.rackspace.com : Resolving timed out after 10000 milliseconds
error: failed retrieving file 'community.db' from mirror.leaseweb.net : Resolving timed out after 10000 milliseconds
error: failed to update community (download library error)
error: failed to synchronize all databases
The command '/bin/sh -c pacman --noconfirm -Syu 	&& pacman --needed --noconfirm -S 		arp-scan 		python 		parted 		dosfstools 		rsync 	&& (pacman --noconfirm -Sc || true) 	&& rm -rf /var/cache/pacman/pkg/*' returned a non-zero code: 1
make[1]: *** [Makefile:197: toolbox] Error 1
make[1]: Leaving directory '/home/pilot/os/.pi-builder'
make: *** [Makefile:54: os] Error 2
`

mdevaev, 2021-01-25:
You have some internet problems on the host (or inside docker)

pawkon, 2021-01-25:
You were right, problem was with the docker but I was unable to solve it (reinstaling etc) so I decided to reinstall Ubuntu (Virtual Machine). I did everything from the begining ant the build has ended properly but now the problem is that it didn't produced image file. I started the image build with `make os NC=1` Should I do now simple `make os` ? I'm attaching full log from make process.
[out_make.txt](https://github.com/pikvm/pikvm/files/5869080/out_make.txt)

mdevaev, 2021-01-26:
The log is fine. Run `make image` now.

pawkon, 2021-01-26:
Okey, there is some progress. I was able to get the w5500 module running. It gets the IP and I'm able to ping it and login on the pikvm www site but there are still some problems. I'm unable to ssh to it. Do I have to change something in the `/etc/ssh/sshd_config` ? Or something else? Because Im getting connection refused

pawkon, 2021-01-27:
It seams that after few restarts ssh let mi in and now everything is working just fine :)

mdevaev, 2021-01-27:
Zero takes a very long time to boot, I think that's the point

pawkon, 2021-01-27:
If someone would like to add ethernet module w5500 from wiznet to pikvm running on raspberry pi zero then here is what I did with help of Maxim:
```
w5500   RPi
Vcc     3v3 (pin1)
Gnd     Ground (pin6)
Mosi    Mosi (pin19)
Sck     sclk (pin23)
Cs      ce0 (pin24)
Int     gpio25 (pin22)
Rst     3v3 (pin17)
Miso    Miso (pin21)
```

There is no need to connect every GND and Vcc because they are connected already on the board.
When the wiring is done then go thru the "Building os tutorial" here : https://github.com/pikvm/pikvm/blob/master/pages/building_os.md
Under the 4th point instead `make os` do `make os NC=1` and when it finishes do `make image`
When it is done clone the image to the sd card and put it in rpi. Run whole system for a while to do the firs boot (it can take even more than 10 minuts)
Then modify the `/boot/config.txt` file and add ad the end:
```
dtparam=spi=on
dtoverlay=w5500
```
You can also add a heat sing to raspberry CPU and add to the config overclocking configuration like this
```
arm_freq=1085
gpu_freq=530
over_voltage=2
core_freq=515
sdram_freq=533
over_voltage_sdram=1

```
above overvoltage parameters taken from here: https://www.raspberrypi.org/forums/viewtopic.php?t=249071
Because this module doen't have a MAC addres it can be set under /etc/systemd/network/00-default.link
```
[Match]
Name=eth0

[Network]
DHCP=yes
DNSSEC=no
MACAddress=xx:xx:xx:xx:xx:xx
```
Now I'm getting 5fps and maybe there can be more tweaking to do ...

Thanks for all help

closed 2021-01-27 by pawkon

mdevaev, 2021-01-28:
Thanks for the detailed description. I've added link to this to the cookbook.

exander77, 2022-03-05:
> `0.0-10.8 sec  7.88 MBytes  6.14 Mbits/sec`

Isn't that kinda slow?

I get `0.00-10.01  sec  5.56 MBytes  4.66 Mbits/sec` with ENC28J60. So, around 30% more from W5500.

pawkon, 2023-03-08:
If You have rainbow loop then there is a problem with the image. Review if any errors while compiling the image and post it here. If not try again from the scratch. Check if the rpi boot up without w5500 module.

marverix, 2023-03-13:
Thanks, @pawkon . For others coming here: As Pi Zero v1.3 is not supported anymore, just use the provided by pikvm image: https://files.pikvm.org/images/v2-hdmi-zerow-latest.img.xz
And use instructions provided by @pawkon on how to connect wires and configure the SPI. Works like a charm! Thanks!