pikvm/pikvm #159: Absolute mouse position buggy under Windows 98

mnotgninnep, 2021-01-23:
**Describe the bug**
Under Windows 98, absolute mouse position (tablet control) is buggy and scaled to 50% of what it should be and is therefore restricted to the top left quarter of the screen. This affects both pi and arduino control. Relative mouse works ok but absolute would be prefered. Windows 98 needs different coordicates (doubled?). This does not appear to be resolution dependent. Control of other OSs is fine.

Virtualbox have documented the same issue here and produced a workaround with a setting change: https://www.virtualbox.org/manual/ch12.html#ts_win98-guest-usb-tablet-coordinates

**To Reproduce**
Steps to reproduce the behavior, like:
Use absolute mouse mode from either Pi or Arduino with a Windows 98 computer.

**Expected behavior**
In absolute mouse mode, the host mouse should exactly follow the client mouse.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 laptop, Windows 98 PC.
 - Browser: Chrome
 - Version: 87.0.4280.141 (Official Build) (64-bit)
 - VNC client: No

**Pi-KVM info:**
  - Raspberry Pi board version Rpi ZeroW
  - Pi-KVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: kvmd 2.13-1
kvmd-platform-v2-hdmi-zerow 2.13-1
kvmd-webterm 0.35-1
  - uStreamer version: ustreamer 3.3-1
  - Linux kernel: Linux pikvm 5.4.83-3-ARCH #1 SMP PREEMPT Mon Jan 4 01:09:25 UTC 2021 armv6l GNU/Linux

mnotgninnep, 2021-01-23:
SilentLennie posted this in discord from VirtualBox which should help. I don't know what to do with it sorry.

SilentLennieToday at 12:02
@mikeyp here is the code which handles the Windows 98 USB tablet correction: https://www.virtualbox.org/browser/vbox/trunk/src/VBox/Devices/Input/UsbMouse.cpp#L211
just look for u8CoordShift in the file
So I guess it's these: https://www.virtualbox.org/browser/vbox/trunk/src/VBox/Devices/Input/UsbMouse.cpp#L1377
https://www.virtualbox.org/browser/vbox/trunk/src/VBox/Devices/Input/UsbMouse.cpp#L1491

mdevaev, 2021-01-23:
Funny bug. I have no chance of getting Windows 98 working right now, but I can try to fix it remotely on your Pi and then make a patch in the release. Contact me at Discord.

mnotgninnep, 2021-01-23:
Awesome, yeah, not worth setting up a Win98 machine. I'll get setup for you to get on. Look for a message from mikeyp. Thank you. :-)

mdevaev, 2021-08-14:
I'm sorry it took so long, but I did it anyway :) The fix for **v2/OTG** will be ready in the next release, I will build it within half an hour. Use these parameters in `/etc/kvmd/override.yaml`:
```yaml
kvmd:
    hid:
        mouse:
            absolute_win98_fix: true
```

closed 2021-08-14 by mdevaev

mdevaev, 2021-08-14:
Also for arduino: https://github.com/pikvm/pikvm/blob/master/pages/arduino_hid.md#fixing-the-usb-absolute-mouse-on-windows-98

mdevaev, 2021-08-14:
@mnotgninnep Let me know if you want dynamic switching between the regular absolute mode and fixed absolute mode in the web UI.

mdevaev, 2021-08-14:
I could do this for Arduino HID, but I need a good reason because the number of available flags in the protocol is limited.

mnotgninnep, 2021-08-14:
I've just built and flashed the image. I'll test it later tonight and get back to you. Thank you again for this. If it works, the option to change it in the GUI would be amazing please. I see no reason to add it to the arduino. Random thought but if you're short of flags, could you manipulate the coordinates before they're sent to the arduino rather than changing the arduino itself?

mdevaev, 2021-08-14:
Unfortunately not. The final coordinates are calculated on the arduino side, this was laid down in the protocol. On the one hand, this ensured compatibility between all firmware versions, but in the case of win98 this became a problem. I'll think about how best to deal with this/

mnotgninnep, 2021-08-14:
Thank you again for this. I can confirm that with the above edit to /etc/kvmd/override.yaml, this works flawlessly. The ability to switch it in the GUI would be greatly appreciated please but I can live with editing it via SSH for now.

jeremyherbert, 2023-06-07:
I am having trouble enabling this feature. I have the top quarter problem on windows 98, and my `/etc/kvmd/override.yaml` file contains the following:

```
kvmd:
    hid:
        mouse:
            absolute_win98_fix: true
```

I have tried a systemctl restart as well as a full reboot, but it doesn't appear to have fixed the problem. Am I doing something wrong here? My version is:

KVMD: 3.199
Streamer: 5.37 (ustreamer)
HAS_PDEATHSIG: Yes
WITH_GPIO: Yes
WITH_PTHREAD_NP: Yes
WITH_SETPROCTITLE: Yes
WITH_SYSTEMD: Yes
Linux kernel:
Machine: armv7l
Release: 5.15.68-3-rpi-ARCH
Version: #1 SMP Mon Oct 31 20:56:54 MSK 2022

mdevaev, 2023-06-07:
@jeremyherbert this code is right here: https://github.com/pikvm/kvmd/blob/master/kvmd/plugins/hid/otg/events.py#L119

Did you try this? https://docs.pikvm.org/mouse/#fixing-the-absolute-mouse-on-windows-98

jeremyherbert, 2023-06-07:
Thank you, I did indeed miss that code. But yes, I did try the adjustment in the docs (my override.yaml is in the earlier post) but it did not work until I hardcoded the shift in.

mdevaev, 2023-06-07:
Please attach the entire override file and output of the `kvmd -m` command.

jeremyherbert, 2023-06-07:
Thanks for your help. My full override file is the lines pasted above - there isn't anything else. Below is the output of that command:

```
ipmi:
    auth:
        file: /etc/kvmd/ipmipasswd

    kvmd:
        timeout: 5.0
        unix: /run/kvmd/kvmd.sock

    server:
        host: '::'
        port: 623
        timeout: 10.0

    sol:
        device: ''
        proxy_port: 0
        select_timeout: 0.1
        speed: 115200

janus:
    check:
        interval: 10.0
        retries: 5
        retries_delay: 5.0

    cmd:
        - /usr/bin/janus
        - --disable-colors
        - --plugins-folder=/usr/lib/ustreamer/janus
        - --configs-folder=/etc/kvmd/janus
        - --interface={src_ip}
        - '{o_stun_server}'
    cmd_append: []
    cmd_remove: []
    stun:
        host: stun.l.google.com
        port: 19302
        retries: 5
        retries_delay: 5.0
        timeout: 5.0

kvmd:
    atx:
        click_delay: 0.1
        device: /dev/gpiochip0
        hdd_led_debounce: 0.1
        hdd_led_inverted: false
        hdd_led_pin: 22
        long_click_delay: 5.5
        power_led_debounce: 0.1
        power_led_inverted: false
        power_led_pin: 24
        power_switch_pin: 23
        reset_switch_pin: 27
        # type: ''
        type: gpio

    auth:
        enabled: true
        external:
            type: ''

        internal:
            file: /etc/kvmd/htpasswd
            force_users: []
            type: htpasswd

[... 351 lines cut ...]
        keepalive:
            count: 3
            enabled: true
            idle: 10
            interval: 3

        max_clients: 10
        no_delay: true
        port: 5900
        tls:
            ciphers: ALL:@SECLEVEL=0
            timeout: 30.0
            x509:
                cert: /etc/kvmd/vnc/ssl/server.crt
                key: /etc/kvmd/vnc/ssl/server.key

    streamer:
        timeout: 5.0
        unix: /run/kvmd/ustreamer.sock

watchdog:
    interval: 30
    rtc: 0
    timeout: 300
```

mdevaev, 2023-06-08:
Ah. I forgot about one thing. Win98 mod depends from relative mouse. Use this:
```yaml
kvmd:
    hid
        mouse:
            absolute_win98_fix: true
        mouse_alt:
            device: /dev/kvmd-hid-mouse-alt
```
After that you will get 3 buttons in the system menu. Switch it to Abs-Win98:
[image]