pikvm/pikvm #160: VNC gives Access Denied errors in the log

redfive2012, 2021-01-23:
When trying to connect to PiKVM via RealVNC, no matter what password I try I get "Invalid password". When I look in the PiKVM log, it tells me "Got access denied for user '<any user name>' from auth service 'htpasswd'". A couple of lines later the log shows "Error: VNCAuth access denied (user not found)". I've tried both "admin" and another one I set up in /etc/kvmd/vncpasswd, same message for both; I even tried using the password for root. I followed the instructions in the VNC page here to enable the service.

TigerVNC gives me the same result, whether I enter both username and password or just the password/passphrase. Given the messages I pasted above from the PiKVM log, it seems like another configuration element is missing, but I have no idea what or where.

**To Reproduce**
1. Open RealVNC Viewer
2. Set up new connection with all defaults
3. Double-click to connect
4. Enter password when prompted

**Desktop (please complete the following information):**
 - OS: Windows 10
 - VNC client: RealVNC Viewer

**Pi-KVM info:**
  - Raspberry Pi board version: V4 B+ 2GB
  - Pi-KVM platform: V2-HDMI
  - Video capture type: CSI bridge
  - KVMD version: 1.100-1, webterm version 0.34-1
  - uStreamer version: 1.25-1
  - Linux kernel: 5.4.61-2-ARCH

mdevaev, 2021-01-23:
RealVNC is not "real" VNC. The VNC protocol contains only basic compression and authorization methods and can't even authorize you by name and password. RealVNC implements its own proprietary extensions for this purpose and does not support open standards. Use, for example, TigerVNC.

redfive2012, 2021-01-23:
TigerVNC gives the same results, with the same errors in the PiKVM log.

mdevaev, 2021-01-23:
Yes, I see, I'm sorry. I just woke up. Perform a full system update, this should help.

redfive2012, 2021-01-23:
pacman -Syu throws "error: failed retrieving file 'pikvm.db' from pikvm.org : SSL certificate problem: certificate is not yet valid" and fails to update.

mdevaev, 2021-01-23:
Probably the system clock can't get to NTP since SSL certificate is valid. Show `timedatectl`.

redfive2012, 2021-01-23:
Yeah, that just occurred to me as well. Updating the clock now.

redfive2012, 2021-01-23:
/etc/systemd/timesync.conf needed to have the NTP= line enabled and a time server configured. Did that and restarted systemd-timesync.service and pacman is updating now.

redfive2012, 2021-01-23:
TigerVNC now connects. RealVNC connects, flashes the screen briefly, and drops the connection, probably due to the compatibility issue you noted earlier. Thanks.

mdevaev, 2021-01-23:
Perfect, glad to help

closed 2021-01-23 by mdevaev