pikvm/pikvm #163: Question about HDMI-to-USB on RPi Zero

pskowronek, 2021-01-27:
I see this statement: `... or HDMI to USB dongle (not available for ZeroW) - high latency ~200ms, not very reliable).`

Why is not available for ZeroW? Is it because of technical limitations or it works poorly due to slow CPU or sth?
I'm looking at this option as I have RPi Zero I could use and I don't need superb quality or high latency.

mdevaev, 2021-01-28:
This is actually available for ZeroW, just requires the use of Arduino for HID emulation, and disqualifies you from using Mass Storage emulation. ZeroW has only one USB port and can be either a host or a device. Also no, you won't be able to use a splitter, hub, or anything like that to get two ports. These are exclusive modes of operation.

pskowronek, 2021-01-31:
Thanks for information!

closed 2021-01-31 by pskowronek