pikvm/pikvm #165: Tailscale fails to start

kchinth, 2021-01-29:
**Describe the bug**
Enabling the tailscale service results in an error:

```
[root@pikvm ~]# systemctl enable --now tailscaled
Created symlink /etc/systemd/system/multi-user.target.wants/tailscaled.service -> /usr/lib/systemd/system/tailscaled.service.
Job for tailscaled.service failed because the control process exited with error code.
See "systemctl status tailscaled.service" and "journalctl -xe" for details.
```

```
[root@pikvm ~]# systemctl status tailscaled.service

* tailscaled.service - Tailscale node agent
     Loaded: loaded (/usr/lib/systemd/system/tailscaled.service; enabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/tailscaled.service.d
             `-override.conf
     Active: activating (start) since Sat 2021-01-30 00:05:37 MSK; 1s ago
       Docs: https://tailscale.com/kb/
    Process: 5326 ExecStartPre=/usr/sbin/tailscaled --cleanup --state=/var/lib/tailscale/tailscaled.state (code=exited, status=0/SUCCESS)
   Main PID: 5342 (tailscaled)
      Tasks: 11 (limit: 3714)
     CGroup: /system.slice/tailscaled.service
             `-5342 /usr/sbin/tailscaled --state=/var/lib/tailscale/tailscaled.state --socket=/run/tailscale/tailscaled.sock --port 41641

Jan 30 00:05:38 pikvm tailscaled[5342]: external route MTU: 1280 (<nil>)
Jan 30 00:05:38 pikvm tailscaled[5342]: Clearing router settings...
Jan 30 00:05:38 pikvm tailscaled[5342]: Starting link monitor...
Jan 30 00:05:38 pikvm tailscaled[5342]: Starting magicsock...
Jan 30 00:05:38 pikvm tailscaled[5342]: Starting resolver...
Jan 30 00:05:38 pikvm tailscaled[5342]: Engine created.
Jan 30 00:05:38 pikvm tailscaled[5342]: Listening on /run/tailscale/tailscaled.sock
Jan 30 00:05:38 pikvm tailscaled[5342]: ipnserver.Run: ipn.NewFileStore("/var/lib/tailscale/tailscaled.state"): open /var/lib/tailscale/tails>
Jan 30 00:05:38 pikvm tailscaled[5342]: flushing log.
Jan 30 00:05:38 pikvm tailscaled[5342]: logger closing down
```

```
[root@pikvm ~]# journalctl -xe

Jan 30 00:07:18 pikvm tailscaled[7314]: Interface set up
Jan 30 00:07:18 pikvm tailscaled[7314]: UDP bind has been updated
Jan 30 00:07:18 pikvm tailscaled[7314]: router: v6nat = true
Jan 30 00:07:18 pikvm tailscaled[7314]: router: dns: using dns.directManager
Jan 30 00:07:18 pikvm tailscaled[7314]: Bringing wireguard device up...
Jan 30 00:07:18 pikvm tailscaled[7314]: Bringing router up...
Jan 30 00:07:18 pikvm tailscaled[7314]: logtail: dialed "log.tailscale.io:443" in 71ms
Jan 30 00:07:18 pikvm systemd-networkd[246]: tailscale0: Link UP
Jan 30 00:07:18 pikvm systemd-networkd[246]: tailscale0: Gained carrier
Jan 30 00:07:18 pikvm systemd-networkd[246]: tailscale0: Gained IPv6LL
Jan 30 00:07:18 pikvm tailscaled[7314]: external route MTU: 1280 (<nil>)
Jan 30 00:07:18 pikvm tailscaled[7314]: Clearing router settings...
Jan 30 00:07:18 pikvm tailscaled[7314]: Starting link monitor...
Jan 30 00:07:18 pikvm tailscaled[7314]: Starting magicsock...
Jan 30 00:07:18 pikvm tailscaled[7314]: Starting resolver...
Jan 30 00:07:18 pikvm tailscaled[7314]: Engine created.
Jan 30 00:07:18 pikvm tailscaled[7314]: Listening on /run/tailscale/tailscaled.sock
Jan 30 00:07:18 pikvm tailscaled[7314]: ipnserver.Run: ipn.NewFileStore("/var/lib/tailscale/tailscaled.state"): open /var/lib/tailscale/tails>
Jan 30 00:07:18 pikvm tailscaled[7314]: flushing log.
Jan 30 00:07:18 pikvm tailscaled[7314]: logger closing down
Jan 30 00:07:18 pikvm systemd-networkd[246]: tailscale0: Link DOWN
Jan 30 00:07:18 pikvm systemd-networkd[246]: tailscale0: Lost carrier
Jan 30 00:07:18 pikvm systemd[1]: tailscaled.service: Main process exited, code=exited, status=1/FAILURE

```

**To Reproduce**
After installing tailscale and running "systemctl enable --now tailscaled", error occurs:
Created symlink /etc/systemd/system/multi-user.target.wants/tailscaled.service -> /usr/lib/systemd/system/tailscaled.service.
Job for tailscaled.service failed because the control process exited with error code.
See "systemctl status tailscaled.service" and "journalctl -xe" for details.

**Expected behavior**
Tailscale service to start.

**Screenshots**
n/a

**Desktop (please complete the following information):**
 n/a

**Pi-KVM info:**
  - Rpi 4
  - v2-hdmi
  - CSI bridge

- kvmd 2.16-1
- kvmd-oled 0.3-1
- kvmd-platform-v2-hdmi-rpi4 2.16-1
- kvmd-webterm 0.36-1

Linux pikvm 5.10.11-4-ARCH #1 SMP Fri Jan 29 18:07:08 UTC 2021 armv7l GNU/Linux

kchinth, 2021-01-29:
Installing the normal tailscale package works fine, so I think it's something with the tailscale-pikvm package and the override.conf.

mdevaev, 2021-01-30:
Previously, this package was required to change the configuration of tailscale as it could not work properly on read only os. Can you confirm that after the reboot, tailscale is working?

kchinth, 2021-01-30:
Yes, I am able to reboot the pi and then connect to tailscale successfully.  Just to test, I rebooted several times and it's still working.

mdevaev, 2021-01-30:
Okay I fixed id. Thanks for the report!

closed 2021-01-30 by mdevaev