pikvm/pikvm #168: Anybody get this to build on a Pi Zero W ?

mocallins, 2021-02-02:
`
Step 2/6 : RUN mkdir -p /etc/pacman.d/hooks     && ln -s /dev/null /etc/pacman.d/hooks/30-systemd-tmpfiles.hook
 ---> [Warning] The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v6) and no specific platform was requested
 ---> Running in f92b669b2196
standard_init_linux.go:219: exec user process caused: exec format error
The command '/bin/sh -c mkdir -p /etc/pacman.d/hooks    && ln -s /dev/null /etc/pacman.d/hooks/30-systemd-tmpfiles.hook' returned a non-zero code: 1
make[1]: *** [Makefile:197: toolbox] Error 1
make[1]: Leaving directory '/home/pi/build/os/.pi-builder'
make: *** [Makefile:54: os] Error 2`

Raspberry Pi Zero W
Raspbian buster
16 GB sd card

mdevaev, 2021-02-02:
This is not intended to work on RPi. Use x86_64 for the build.

mocallins, 2021-02-03:
> This is not intended to work on RPi. Use x86_64 for the build.

What does this mean ?  This is the Pi-KVM project correct ?

joBr99, 2021-02-03:
Yeah, but you have to build it on a x86_64 computer, in that process you are flashing it to an sd card which you can plug into your raspberry.

For Raspberry Pi 4 you can download a pre-build image from https://pikvm.org/download.html

mocallins, 2021-02-04:
'''
Step 23/88 : RUN echo "Server = $REPO_URL/\$arch/\$repo" > /etc/pacman.d/mirrorlist     && pacman-key --init    && pacman-key --populate archlinuxarm
 ---> Running in 524b57bc716b
gpg: /etc/pacman.d/gnupg/trustdb.gpg: trustdb created
gpg: no ultimately trusted keys found
gpg: starting migration from earlier GnuPG versions
gpg: porting secret keys from '/etc/pacman.d/gnupg/secring.gpg' to gpg-agent
gpg: migration succeeded
gpg: Generating pacman keyring master key...
gpg: agent_genkey failed: End of file
gpg: key generation failed: End of file
gpg: Done
==> Updating trust database...
gpg: no need for a trustdb check
==> Appending keys from archlinuxarm.gpg...
gpg: key B55C5315DCD9EE1A was created 16088 days in the future (time warp or clock problem)
gpg: key B55C5315DCD9EE1A was created 3329 days in the future (time warp or clock problem)
'''
My goodness, how long does it take to build this ?

rqdev14, 2021-02-05:
So I have been running into similar issues for the raspberry pi zero w. Setup done using Ubuntu 20.04. 

I assume there is a typo in the guide for Wi-Fi, using 
`WIFI_SSID = "my-network"
WIFI_PASSWD = "P@$$word"`
instead of
`WIFI_ESSID = "my-network"
WIFI_PASSWD = "P@$$word"`
helped me 'make os'

However I am still getting errors when I run 'make install'. 
To work around this, I used 'make image' which resulted in a .bz2 file, unfortunately this file is corrupted and fails to write. After checking, this file results in a checksum error. 

Is anyone else facing this issue?/Is there any work around?

mocallins, 2021-02-05:
I've tested changing the WIFI related values and even commenting them out, still locks up at the same step  23/88

mdevaev, 2021-02-05:
@mocallins which os do you use?

mocallins, 2021-02-05:
Debian 10.06

mdevaev, 2021-02-05:
The errors you get indicate problems with glibc / kernel. Perhaps some of this is too old (there is a problem with one system call returning the wrong time). Use a VM running ubuntu 20.04 to build.

mocallins, 2021-02-05:
Ok, i actually have a Ubuntu 20.04, will  try there.

mocallins, 2021-02-05:
'''
===== Ensuring base rootfs =====
if [ ! -e ./.tmp/base-rootfs-zerow.tar.gz ]; then \
        mkdir -p ./.tmp \
        && curl -L -f http://de3.mirror.archlinuxarm.org/os/ArchLinuxARM-rpi
est.tar.gz -z ./.tmp/base-rootfs-zerow.tar.gz -o ./.tmp/base-rootfs-zerow.ta
 \
; fi
Warning: Illegal date format for -z, --time-cond (and not a file name).
Warning: Disabling time condition. See curl_getdate(3) for valid date syntax
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Curr
                                 Dload  Upload   Total   Spent    Left  Spee
  0     0    0     0    0     0      0      0 --:--:--  0:00:09 --:--:--
rl: (6) Could not resolve host: de3.mirror.archlinuxarm.org
make[1]: *** [Makefile:274: _rpi_base_rootfs_tgz] Error 6
make[1]: Leaving directory '/home/mike/build/os/.pi-builder'
make: *** [Makefile:54: os] Error 2'''

Aaarrgghh, now i remember why i used Debian

mdevaev, 2021-02-05:
> Could not resolve host: de3.mirror.archlinuxarm.org

I seems you have some network problems

mocallins, 2021-02-05:
Yea and i have no idea what to do

mdevaev, 2021-02-05:
Go to discord, I think someone will help you soon there. If they don't help, I'll build the image for you myself.

mocallins, 2021-02-05:
Uuugghh, yea i'm not joining discord

mocallins, 2021-02-05:
@mdevaev  its weird, i can get the command to run manually, so its not a routing problem.
But since it gets crashed at 9 seconds, and it takes my system about 13 seconds to get the download stated, kinda weird.

Do the scripts expect a default shell ?
Becuase i use tcsh, and have tons of aliases setup.

mdevaev, 2021-02-06:
It doesn't matter. The build runs inside docker, so if it can't get to the internet, then something is wrong with the docker network.

mocallins, 2021-02-06:
Hmmm, well i have no practical experience with docker.

mocallins, 2021-02-06:
I do see that i have a docker0 network interface witha 172.17.0.1  address. 
can ing it.
No idea what to do to troubleshoot anything with docker.

mdevaev, 2021-02-06:
Let me just build you an image myself. You just get unlucky with this - a lot of third-party bugs per square meter. Contact me at Discord.

Testato, 2021-02-06:
Why not put in Download section also an RpiZeroW image ?
Rpi4 and Pizero is the most used i think

mdevaev, 2021-02-06:
@Testato You need to be able to configure the network. Now this is not available, only with the help of the assembly system.

bstin, 2021-02-10:
@mdevaev  - Wouldn't it be possible to just boot pi-zero with keyboard / monitor attached and set wifi credentials from arch command-line? 

This seems much easier than to build a new image for every unit?

mdevaev, 2021-02-10:
I plan to make a network setup service at boot, this will solve the problem

mdevaev, 2021-02-13:
Try the ready-made image
https://discord.com/channels/580094191938437144/742948323387899934/809840883629424751

mocallins, 2021-02-13:
I'll try it.

mdevaev, 2021-02-15:
The finished image is now the official way to use Zero. The instructions have also been updated:
* https://pikvm.org/download.html
* https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md

closed 2021-02-15 by mdevaev