pikvm/pikvm #169: Support for PoE

dkrutsko, 2021-02-04:
I just found out about this project and I'm getting really excited about its potential uses. One consideration though is documenting and potentially creating a PoE variant of the top hat.

You see, in a rack-mounted environment, I could envision four or more of these Pi-KVMs being slotted and connected to their own machines. Ideally I would want the wiring to be as simple as possible. Having power be delivered alongside the ethernet cable would remove the need for a whole wire and power solution to be installed within the rack.

However, in the meantime, are you able to document the requirements for a third-party PoE solution which would be compatible with version 3 of the board. Specifically, power requirements and other considerations. It would also be nice to have some recommendations for existing PoE solutions which would be compatible.

mdevaev, 2021-02-06:
Hi! We are creating v3 so that it can be used with PoE. It seems that all existing boards are compatible with this (except for the official RPi PoE, it requires turning off the fan since it conflicts with the HDMI chip, but this is not a big problem). My only advice is that you need an honest PoE with a transformer. This is more reliable.

dkrutsko, 2021-02-06:
I've been seeing a lot of PoE hats which can deliver 15 watts of power. Is this going to be enough for the PiKVM, or does it require further testing?

mdevaev, 2021-02-07:
I think yes

closed 2021-02-14 by mdevaev

cipy, 2022-01-18:
hi @mdevaev 

would (any of the) existing PoE HATs fit inside your custom PiKVM HAT v3 metal case? 🤔 

i.e. from https://thepihut.com/pages/search-results?q=power%20over%20ethernet, thanks

mdevaev, 2022-01-18:
the existing ones won't fit, but we plan to release our poe module by the summer

alichaudry, 2022-08-21:
> the existing ones won't fit, but we plan to release our poe module by the summer

Hi @mdevaev, are there any updates on the PoE module as we approach the end of summer? I require PoE for my needs and would like to go down the PiKVM route rather than the alternatives that provide PoE.

mdevaev, 2022-08-21:
Hello. We are working on it. There are no updates right now.

unknownusermello, 2023-01-27:
hey I know that you guys had the Kickstarter going on but I was wondering if there are any updates for the PiKVM V3 PoE addon that you mentioned back in February 2021?

Gooberpatrol66, 2023-01-27:
does the v4 have PoE?

alichaudry, 2023-01-27:
> does the v4 have PoE?

No, from what I read on [this thread](https://www.reddit.com/r/HomeServer/comments/znj4cm/pikvm_v4_nextgen_kvm_over_ip_on_kickstarter/) on reddit, pikvm v4 does _not_ have PoE. Pretty much the only reason I didn't pull the trigger and buy it (I absolutely love the new design of the PiKVM v4 mini otherwise).

thomasdiemarbennedsen, 2023-08-08:
Im also waiting for PoE support for a version that fits a 1u installation for instance https://www.waveshare.com/pi4-1u-rack-kit.htm

What is the status for PoE support?

mdevaev, 2023-08-08:
We have a prototype now, and just choosing a common chip for all V3 and V4 models.

Gooberpatrol66, 2023-08-08:
> Im also waiting for PoE support for a version that fits a 1u installation for instance https://www.waveshare.com/pi4-1u-rack-kit.htm
> 
> What is the status for PoE support?

I've never seen a pi rackmount that actually looks like it'd fit a pikvm. It needs holes cut out for the ports on the hat. I'm hoping to just mount my pikvm's on a DIN rail.

starsoccer, 2024-05-14:
Any update on this?