pikvm/pikvm #170: Sync Macro execution?

GregoryOrciuch, 2021-02-04:
**Is your feature request related to a problem? Please describe.**
When you run pikvm and chrome connected to it's web interface on computer A there you run macro e.g. moving mouse in infinite loop. Then you change your location and use computer B and access same webpage of pikvm, you can see the mouse moving cause computer A macro moves it, it looks like computer's A macro cant be stopped from another web interface instance.

**Describe the solution you'd like**
Sync macro state via pikvm backend ? Option for stop another running macro ?

**Describe alternatives you've considered**
Maybe that rises a flag for some multiuser access for same hdmi ?

**Additional context**
n/a

mdevaev, 2021-02-06:
Scripts are now executed entirely on the frontend. In the future, we can make an API for the backend.

Majestic7979, 2025-02-06:
I would also greatly benefit from server-side running of macro...

It has been done for the mouse jiggler function, so expanding would be amazing.

Every day I have to juggle between two workstations, I need a macro for a specific reason to keep one specific device running (it's a laptop, and it tends to fall asleep despite energy settings, due to bad bios I guess - I tried everything but only moving mouse or pressing a keyboard key periodically keeps it running).

The problem is, if the macro is already running, then I get some unwanted behavior that would be totally resolved if I could manage the macro server-side.

So my need/requirement includes the ability to upload a macro to PiKVM and have it there always. It would be great if there was a "macro editor", macro storage server-side, and running/stopping/looping(very important) directly on the server. That would make this solution perfect for me. I also wish that the ATX Menu would support virtual power button via WOL or another method, because it's a laptop there's no way to install the ATX Addon card. I even tried the crazy method of opening it up and soldering wires to the power switch (unsuccessfully) because I thought I could create a GPIO item with a relay to "press" the power virtually.

Important to know: This is a corporate device, I cannot change anything on it, or the device itself, so PiKVM is used as a workaround regarding the limitations.