pikvm/pikvm #171: TESmart 8-port Script Question

taylormia, 2021-02-06:
Thanks for this awesome piece of software. 
I'm planning on getting the TESmart 8-port KVM switch to use with pikvm. 
I gather from the readme that I will have to upload the [kvmctl.sh](https://github.com/bbeaudoin/bash/tree/master/tesmart) script using the pikvm web UI after customizing the IP address in the script and connecting the Ethernet port and setting a fixed IP address on the KVM switch. Is this correct?
What should I expect to see in the web UI after uploading the script?

mdevaev, 2021-02-06:
The interface does not display anything for this switch, you need to use the terminal to change channels.

taylormia, 2021-02-06:
Would I have to invoke the kvmctl.sh script from the pikvm terminal every time I needed to switch the port? What's the process to switch ports from the terminal?

mdevaev, 2021-02-11:
Yes. I don't know for sure, this switch was tested not by me but by one of the users in Discord. In principle, nothing prevents you from making a pure python plugin for KVMD as for the Ezcoo switch: https://github.com/pikvm/pikvm/blob/master/pages/ezcoo.md

Then you will be able to control it from the interface.

taylormia, 2021-02-12:
I'm not a coder...so would not know how to make a python plugin for the TESmart switch.....If you are referring to the override.yaml...I can do that but not sure how to find the values for each channel etc.

mdevaev, 2021-02-12:
I see. No, one config is not enough here. Go to our discord and write to the user **baddog**, it was he who tested this switch.

I will probably write a plugin in the future, but unfortunately I'm critically short of time right now.

taylormia, 2021-02-12:
Will do. Thanks

bbeaudoin, 2021-02-18:
That user is me btw, @bbeaudoin. I may have broken the link to my writeup and script on the KVM and am interested in writing a plugin, I just haven't scoped out the level of effort yet but I don't believe it's very high.

When I have some dedicated time to look at this I'll do that.

taylormia, 2021-02-18:
Thanks @bbeaudoin . Appreciate your effort on this.

mdevaev, 2021-03-01:
I suppose I can close this issue.

closed 2021-03-01 by mdevaev