pikvm/pikvm #172: error: failed to initialize alpm library (could not find or read directory: /var/lib/pacman/)

btafoya, 2021-02-07:
I am trying to build an image for the pi zerow on Ubuntu 20.04 and get only so far. I have attempted to build on two different systems with the very same result.

Below is the output from the build:

git clone --depth=1 https://github.com/mdevaev/pi-builder ./.pi-builder
Cloning into './.pi-builder'...
remote: Enumerating objects: 50, done.
remote: Counting objects: 100% (50/50), done.
remote: Compressing objects: 100% (39/39), done.
remote: Total 50 (delta 0), reused 34 (delta 0), pack-reused 0
Unpacking objects: 100% (50/50), 3.98 MiB | 1.13 MiB/s, done.
rm -rf ./.pi-builder/stages/{pikvm,pikvm-image,pikvm-otg-console}
cp -a pikvm pikvm-image pikvm-otg-console ./.pi-builder/stages
make -C ./.pi-builder os \
        NC= \
        BUILD_OPTS='  \
                --build-arg PLATFORM=v2-hdmi \
                --build-arg USTREAMER_VERSION=3.16-1 \
                --build-arg KVMD_VERSION=2.17-1 \
                --build-arg KVMD_WEBTERM_VERSION=0.36-1 \
                --build-arg WIFI_ESSID="somewifi" \
                --build-arg WIFI_PASSWD="somewifipass" \
                --build-arg WIFI_IFACE=wlan0 \
                --build-arg ROOT_PASSWD=root \
                --build-arg WEBUI_ADMIN_PASSWD=admin \
                --build-arg IPMI_ADMIN_PASSWD=admin \
                --build-arg NEW_HTTPS_CERT=a005031f-857e-4cc2-8fb5-98bf1da97cc3 \
        ' \
        PROJECT=pikvm-os-v2-hdmi \
        BOARD=zerow \
        STAGES='__init__ os pikvm-repo watchdog ro no-audit pikvm pikvm-image __cleanup__' \
        HOSTNAME=pikvm \
        LOCALE=en_US \
        TIMEZONE=America/New_York \
        REPO_URL=http://de3.mirror.archlinuxarm.org
make[1]: Entering directory '/home/ubuntu/os/.pi-builder'
===== Ensuring toolbox image =====
docker build \
                --rm \
                --tag pi-builder-arm-toolbox \
                 \
                --file toolbox/Dockerfile.root \
        toolbox
Sending build context to Docker daemon  15.36kB
Step 1/6 : FROM archlinux/archlinux:base-devel
base-devel: Pulling from archlinux/archlinux
4e2e2d69f02e: Pull complete
c2c608c60eeb: Pull complete
Digest: sha256:bbc36ffb843a3a542d6150eb3f5ef0a4cd5d14d9d1ff51a37ad5dcbf5ecec506
Status: Downloaded newer image for archlinux/archlinux:base-devel
 ---> bb1d16ed17e2
Step 2/6 : RUN mkdir -p /etc/pacman.d/hooks     && ln -s /dev/null /etc/pacman.d/hooks/30-systemd-tmpfiles.hook
 ---> Running in 1155a134fe63
Removing intermediate container 1155a134fe63
 ---> 1ef62ba9ddd3
Step 3/6 : RUN pacman --noconfirm -Syu  && pacman --needed --noconfirm -S               arp-scan                python          parted          dosfstools              rsync   && (pacman --noconfirm -Sc || true)     && rm -rf /var/cache/pacman/pkg/*
 ---> Running in 7805eff2a43a
error: failed to initialize alpm library
(could not find or read directory: /var/lib/pacman/)
The command '/bin/sh -c pacman --noconfirm -Syu         && pacman --needed --noconfirm -S               arp-scan                python          parted          dosfstools              rsync   && (pacman --noconfirm -Sc || true)     && rm -rf /var/cache/pacman/pkg/*' returned a non-zero code: 255
make[1]: *** [Makefile:197: toolbox] Error 255
make[1]: Leaving directory '/home/ubuntu/os/.pi-builder'
make: *** [Makefile:54: os] Error 2

mdevaev, 2021-02-07:
Strange problem. I can't play it locally, but it plays on another user too. I'll figure it out.

mdevaev, 2021-02-07:
Here the problem: https://bugs.archlinux.org/task/69563

btafoya, 2021-02-08:
Gotcha, bummer! Any suggestions to work around the glibc issue?

mdevaev, 2021-02-08:
Try to remove the `os` and clone it again. There is a small fix in the repository.

btafoya, 2021-02-08:
Thank you very much!

btafoya, 2021-02-08:
One more issue related to glibc 2.33:

===== Ensuring arm binfmt =====
docker run \
                --rm \
                --tty \
                --privileged \
        pi-builder-arm-toolbox /tools/install-binfmt \
                --mount \
                arm \
                /usr/bin/qemu-arm-static
python3: /usr/lib/libc.so.6: version `GLIBC_2.33' not found (required by /usr/lib/libpython3.9.so.1.0)
make[1]: *** [Makefile:208: binfmt] Error 1
make[1]: Leaving directory '/home/ubuntu/os/.pi-builder'
make: *** [Makefile:54: os] Error 2

mdevaev, 2021-02-08:
Oh crap. Try to create a virtual machine with arch linux and build the image there. On my arch, this is going to happen.

btafoya, 2021-02-08:
Success on arch! Thank you again for your help and this awesome project!

mdevaev, 2021-02-08:
Great! Let's leave this issue open so that others can see this solution, however temporary.

alexsalex, 2021-02-12:
Same issue on ubuntu 20.04 and 20.10

mdevaev, 2021-02-16:
I have updated the host system requirements in the documentation, there is nothing more I can do here. This part is about glibc, so either use ready-made images from my site, or you need a virtual machine with Arch Linux. When the rest of the OS updates docker and the system libraries, it will work again.

closed 2021-02-16 by mdevaev

db260179, 2021-02-27:
Just to add to this, the current Debian testing iso now builds these images (once docker engine is installed selecting the debian 10 version)

mdevaev, 2021-02-27:
Perfect