pikvm/pikvm #179: Build system other than Archlinux

db260179, 2021-02-11:
**Is your feature request related to a problem? Please describe.**
Current build system using Arch linux is very problematic and breaks alot

**Describe the solution you'd like**
What steps or changes or limitations to use Debian as a base system?

**Describe alternatives you've considered**
Use a base build system on Ubuntu or Debian stable

Im happy to assist in getting this to work.

As you are using Docker to build the code, has there been any attempt to use Ubuntu or Debian as a base system.

db260179, 2021-02-11:
Arch linux is a great next gen features OS, but its rapid pace means alot of random breakages and lots of bugs.

I can envisage the following is needed to be done:

1. Identify the packages and dependencies
2. Test build code
3. Identify issues based on versions of packages used

mdevaev, 2021-02-11:
For the Debians, there will be too much to change. Many packages need to be packaged and tested. I'm not against the alternative, but right now I don't have the resources to do it.

alexsalex, 2021-02-12:
Would like to help

db260179, 2021-02-12:
I understand, im happy to make a start. If you are ok with that, can I ask questions later if I get stuck with your packaging?

The docker parts are straightforward, the packages and dependencies is the tricky bit.

I will update you on progress. This is my way of contributing.

mdevaev, 2021-02-15:
Ofc you can contact me via discord or here.

mdevaev, 2021-07-14:
Related: #245

mdevaev, 2021-07-16:
I'm closing it since it seems that we have those who are going to do this in another issue.

mdevaev, 2021-07-16:
See #374

closed 2021-07-16 by mdevaev