pikvm/pikvm #180: permission denied on /etc/kvmd/nginx/ssl/ when starting kvmd

thegogglesdonothing, 2021-02-12:
I compiled the code in a clean Arch Linux vm build environment.

When booting up the pi, attempting to access the console via the browser returns a nginx 500 internal server error page.

The kvmd service fails to start with a permission denied on the ssl certificate file

/usr/lib/systemd/system/kvmd*.service files list kvmd:kvmd as the service user

but permissions on the ssl directory are root:kvmd-nginx and the kvmd user is not a member of the kvmd-nginx group so the nginx service fails to start.

chown kvmd on the ssl directory, or adding the kvmd user to the kvmd-nginx group fixes the problem.

mdevaev, 2021-02-12:
Yes, my fault. It's already fixed. Just perform update: `rw; pacman -Syu; reboot`

thegogglesdonothing, 2021-02-13:
Recompiled from start to finish, works perfectly now, thanks!

closed 2021-02-13 by thegogglesdonothing