pikvm/pikvm #181: Pi Zero W - WiFi doesn't start or connect

alexsalex, 2021-02-13:
I deployed the Arch Linux VM (only arch works for me, Ubuntu doesn't work). In Arch VM created the image to install it on RasPi Zero W. After I burn it on the MicroSD (minimum 8GB need, because the image is more than 6Gb !?), put it in the RasPi.... and nothing.

My WiFi have space in the name, but I check the Arch support, and it shouldn't be the issue. I can see my network config in /etc/netctl, looks like everything looking good, but can't connect to the WiFi, do not see the device in my network. 

My WiFi have the same name for 2.4 and 5 GHz. But based on the Arch forum conversation it shouldn't be the issue too.

How can I debug the WiFi or check the logs? New toArch Linux.

Do not have the keyboard for RasPi Zero... so, can't check the working system, but can check the micro SD.

Thank you!

mdevaev, 2021-02-13:
Hi. Here the problem: https://archlinuxarm.org/forum/viewtopic.php?f=15&t=15120

TLDR broke wifi support and poke fingers at each other.
[image]

**Solution**
1. Download the ready-made image from the site: https://pikvm.org/download.html
2. Follow the updated guide: https://github.com/pikvm/pikvm/blob/master/pages/flashing_os.md

alexsalex, 2021-02-13:
> 1. https://pikvm.org/download.html

You have ONLY for Pi4... I use Raspberry Pi Zero W. Is it ok to install that image on Pi Zero?

mdevaev, 2021-02-13:
Clear the browser cache and try again.
[image]

alexsalex, 2021-02-13:
> Clear the browser cache and try again.
> [image]

Got it) Thank you!

alexsalex, 2021-02-13:
**After installation, we recommend you to update your operating system**

Should I?..

mdevaev, 2021-02-13:
Yes

alexsalex, 2021-02-13:
Looks good for me. Really appreciate your help! Спасибо)

But still... Arch is pain in ass.... Will be much better to have Debian based bistro. Thank you again)

closed 2021-02-13 by mdevaev