pikvm/pikvm #182: pikvm destroys pi zero

jonasled, 2021-02-13:
**Describe the bug**
I and a friend had a pikvm running on a pi zero. Both systems are now dead. The pi is getting very hot and doesn't boot. It looks like there is a bug which kills the device.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Install pikvm on a pi zero
2. keep it running for a few weeks

**Expected behavior**
Pi works

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**Pi-KVM info:**
  - zero W
  - usb2HDMI and the other had CSI 
  - one had the pi usb and the other one a pro micro
  - KVMD version: SD card dead (burned by pi), was build on 18.09.20

mdevaev, 2021-02-13:
Do you have any direct data that confirms this issue? For example, you were monitoring the CPU load all this time or something like that. My software doesn't do anything that could fry the CPU even theoretically. But perhaps the reason lies in the kernel.

mdevaev, 2021-02-13:
I'm really sorry about your Pi. Let's get this sorted out so that no one else has any more problems like this. I'm surprised because out of thousands of users, no one has ever reported something like this. And I test my own changes.

jonasled, 2021-02-13:
No, I dont have any logs from this time. I use the device only once in a while for debugging and suddenly it stopeed working. After I opened the case, the Pi was very hot (it has a cooler on it) and wasn't booting anymore) I tried it a few times more and now also the micro SD is fried.

BrokoliMC, 2021-02-13:
My Pi does the same thing (I am the friend that jonas mentioned above).
I dint do any monitorig of the CPU, i just removed and plugged back in the power adapter of the pi (2A psu).
As jonas described, my pi also died completely but the SD-Card still works.
Maybe there are some kind of logs on there, i dont know if pikvm writes any logs.
I also had the breadboard power thing plugged into the pi and it worked perfectly until everything died.

mdevaev, 2021-02-13:
@jonasled Can you try a different SD card and a regular Raspbian image? The Raspberry chip is very strong and may fail in the memory card.

@BrokoliMC Make an image of the memory card using dd and send it to me. I'll try to toast my Zero with it to honor the demise of your devices.
```
dd if=/dev/mmcblk0 of=zero.img
bzip2 zero.img
```
Upload resulting zero.img.bz2 to some file hosting.

jonasled, 2021-02-13:
I tried already a different card, but the pi is dead.

BrokoliMC, 2021-02-13:
@mdevaev Ill upload an zipped image shortly

jonasled, 2021-02-13:
I can also upload a image, I tried the card on a different device and it looks like it's working

mdevaev, 2021-02-13:
@jonasled Can you give me ssh to this device?

jonasled, 2021-02-13:
I haven't booted the system, I just plugged it into a PC

mdevaev, 2021-02-13:
Ah, okay. Then you will also remove the image. My HW engineer says that burning the Pi due to the operating system is extremely unlikely, as the Pi does throttling under heavy load. He's very curious about what exactly broke, so I'm offering you both a small deal.

Send us your dead zeros by mail, and we will buy you new ones (or send the money to paypa)l. We are sure that the software is not to blame, but we are very interested in doing an autopsy. How do you see it?

By the way: did you buy these devices at the same time? Are they from the same party?

BrokoliMC, 2021-02-13:
My pi is about an year old and we didnt buy them at the same time, i perfectly fine with sending it in

jonasled, 2021-02-13:
Yes, I can send you the pi. It's from about 2017 (shortly after release)

mdevaev, 2021-02-13:
Excellent. We'll do an X-ray, maybe saw through the chip, and find out what happened. If this is a hardware problem and throttling is not working, then we will put the Raspberry Pi on notice.

Contact me at Discord, I'll give you the address. If you live nearby, you can send one package.

mdevaev, 2021-02-13:
It would also be great to see photos of your devices and a detailed description of the builds.

jonasled, 2021-02-13:
The pi looks a bit abused, because it was used before for other projects. It has a arduino pro micro attached for USB and a USB Capture card for the HDMI signal

[image]
[image]

BrokoliMC, 2021-02-13:
Here are some Pictures of my Pi with the hdmi bridge, i currently dont have my breadboard with the modules on it, but i did it exactly like the scematic
[image]
[image]
[image]
[image]
[image]

mdevaev, 2021-02-15:
I tried your images and didn't see anything strange. Well, the autopsy will show.

gsmith28, 2021-03-03:
I'm experiencing the same issue when flashing with the pre-built image for the Pi ZeroW with CSI/HDMI.  The board is accessable for a period (albeit still battling with HID issues) and then it got extremely hot (almost burned my fingers) when I tried pluggin in external power.  My board does NOT appear to be dead; and appears functional (still HID issues) after I left it to cool for a few hours.  Wondering if there is some type of issue with the pre-built image that is causing this hardware fault?

mdevaev, 2021-03-03:
@gsmith28 Run htop on your pi, let's see if there is anything there that loads the CPU. There is an opinion that perhaps this is somehow related to a bug in rngd, which is present in the distro upstream (excessive CPU load). I recently made a workaround for this.

But in fact, I very much doubt that this is a related problem, because the raspberry has throttling and must reduce the frequency in order to avoid overheating.

BrokoliMC, 2021-03-04:
Maybe its because of the Power from the PC via the OTG-Cable gets in the PI in addition to the normal PSU on the Only-Power port.
I will test my new PI Zero with a 2 Amp PSU on the Only-Power port and ill cut the Powerwires on the OTG-Cable, so that the PI doesnt get accidental Power from the PC its connected to.

mdevaev, 2021-03-05:
I don't think it can be related. Raspberry has only one power bus, which just connects two lines. Moreover, they share a common ground.

animeai, 2021-03-07:
Interesting that both dead Pi-Zeros have self-soldered headers. I would suspect a soldering mishap, potentially a solder bridge direct to ground from a current carrying pin. There is also a lot of burning on the solder joints which could have also damaged the board. Will be interested to see the results of the autopsy.

mdevaev, 2021-03-07:
One of the boards was delivered. I'll wait for the second one and send them for an X-ray.

BrokoliMC, 2021-03-07:
> Interesting that both dead Pi-Zeros have self-soldered headers. I would suspect a soldering mishap, potentially a solder bridge direct to ground from a current carrying pin. There is also a lot of [...]

My board worked Flawlessly for propably 2 months or so

mdevaev, 2021-07-14:
We are continuing our research. But the fact that no one else has reported such problems to us makes us think about a defect in the crystal. I recently had a case with RPi4, which required increased power to the video core, otherwise it would hang. That is, it is quite real.

<details>
  <summary>X-ray</summary>

[image]
[image]
[image]
[image]
</details>

mdevaev, 2021-07-28:
Okay, I should have checked it right away. Zero by @jonasled turned out to be working, but its UART was burned. We were able to load the OS and force it to connect via Wi-Fi to our router.

Yes, indeed, you observed overheating due to a working rngd, I will fix it. But heating up to 60 degrees is not critical, the device can live with it (as observed). Most likely, the UART port was burned due to an incorrect Arduino circuit or something. Anyway, the console is not available and does not pass data through itself.

The OTG port is working properly and emulates the USB gadget as it should. The only problem is that it is very loose and has a bad contact.

We didn't get the second device, but most likely the same problem is with it.

bbeaudoin, 2021-07-30:
I started testing this in March 2021 using kvmd-2.29-1. After a month, I was unable to reproduce the issue so I moved the Pi-KVM to another host on April 6, 2021 and left this running for 114 days. I was not able to observe a failure. `rngd` settled after about 24 hours (once it had refilled the entropy pool).

```
[root@pikvm ~]# pacman -Q --info kvmd
Name            : kvmd
Version         : 2.29-1
Description     : The main Pi-KVM daemon
Architecture    : any
URL             : https://github.com/pikvm/kvmd
Licenses        : GPL
[...]
Build Date      : Mon Mar 1 09:47:52 2021
Install Date    : Mon Mar 1 11:55:43 2021
[...]
```

I only set out to examine it after I saw the physical unit was sent in and the hardware was checked over. Here's the running information from my system.

```
[kvmd-webterm@pikvm ~]$ uptime
 16:55:06 up 114 days, 21:19,  0 users,  load average: 0.62, 0.48, 0.52
[kvmd-webterm@pikvm ~]$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 797.66
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 000000004ca0cad3
Model           : Raspberry Pi Zero W Rev 1.1
[kvmd-webterm@pikvm ~]$ ps -aef|grep rngd
root       232     1  0 Apr06 ?        00:04:38 /usr/bin/rngd -f -o /dev/random -r /dev/hwrng -x 5
kvmd-we+  8813  8775  0 16:55 pts/0    00:00:00 grep --color=auto rngd
[kvmd-webterm@pikvm ~]$
```

Over the past 114 days, `rngd` had been running for 114 days but only executed for 4 minutes 38 seconds of CPU time. All of the features continued working throughout the time the system had been running, no software updates or configuration changes were applied to the system manually (only through the automated firstboot configuration).

I could not reproduce the issue, perhaps it was the microcontroller soldered to the UART that resulted in an issue. It's difficult to tell from the photos but it appeared +5V was connected to the microcontroller with UART soldered to the 3.3V GPIO. Both the potential for damage via backpowering and overvoltage on the 3.3V UART likely contributed to failure.

mdevaev, 2021-07-30:
@jonasled @BrokoliMC boop. Do you think there could be an error in the circuit with Arduino?

@bbeaudoin It's definitely not backpowering because it doesn't touch RXTX.

bbeaudoin, 2021-07-30:
@mdevaev backpower speculation due to +5V connected from the Pi Zero W to a device with USB OTG that was possibly connected to a USB port. As for RXTX it's difficult for me to tell given I don't know that microcontroller board and it's still difficult to see from the pictures, but would explain the UART being dead. It's also possible there was a physical short of the pins that would not appear in the pictures and no longer exists.

jonasled, 2021-07-30:
Maybe it was a problem with the uart on my pi. I just used a voltage divider, to reduce the coltage feeded into the pi. I measured the voltages a few times and everythink looked great. @BrokoliMC on the other hand used the otg function, so there was no arduino connected. Interesting is also, that the pi is working again. Because I can't get it to boot with pikvm I also tried a fresh raspbian and had also no luck booting it up.

mdevaev, 2021-07-30:
We did not get a second device to check this due post problems, but now it is clear that rngd could not cause a burning of raspberry. Perhaps there was a short circuit on your board that was eliminated while you were disassembling it or while we were trying to turn it on. It seems to me that if there really was some kind of software defect, there would be much more deaths.

Stranger things.

bbeaudoin, 2021-07-30:
It's hard to say in retrospect. Given my unit has been running for just 6 days short of 4 months continuous uninterrupted operation in a fully enclosed case without a heatsink, I will now be upgrading the unit to kvmd-3.11-1 along with the latest software updates from Arch Linux ARM.

During the package upgrade I did see the temperatures go to 61.6° C, post-install the temps dropped quickly. This is well within operational temperatures (0°-70° C) documented at https://www.raspberrypi.org/documentation/faqs/#pi-performance-temps.

BrokoliMC, 2021-07-30:
> @jonasled @BrokoliMC boop. Do you think there could be an error in the circuit with Arduino?

As @jonasled said, i was not using the Arduino, but the OTG-Cable, i can imagine, that the Power from the PC-USB-Port and the Power from the USB-PSU may have caused some sort of voltage thing (idk how to say it)

I tried to Send the PI again, but the Package seems to be lost again...

My new Pi Zero W with the Same circuit, but with cut Power lines on the OTG-Cable works perfectly for the last 2 Months

mdevaev, 2021-08-01:
Also RNGD broke the entropy generation sources. Due to the update of rng-tools, in some cases, there may be a large consumption of the CPU and its overheating. To avoid this, after updating the system, change the generator parameters and reboot: `echo 'RNGD_OPTS="-o /dev/random -r /dev/hwrng -x jitter -x pkcs11 -x rtlsdr"' > /etc/conf.d/rngd`

opiedopy, 2021-08-10:
My Raspberry Pi zero W, headless, was powered on for Putty ssh testing only for 14 continuous days, no solder, no camera, just power and ssh connection.  It died today. Very warm.

mdevaev, 2021-08-10:
What OS image did you use?

bbeaudoin, 2021-08-10:
This is not unique to Pi-KVM on the Pi Zero W:

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles-.../raspberry-pi-zero-w-died/

I checked my other Pi Zero W, it's fine but has a heatsink. The one that is fully enclosed with the CSI-2 capture board isn't generating appreciable heat at the moment, it's been running over 120 days and was rebooted last week to apply the latest updates, including the new `rngd` configuration:

```
[root@pikvm ~]# uptime
 17:43:01 up 6 days, 17:18,  1 user,  load average: 0.50, 0.60, 0.57
[root@pikvm ~]# pistat
# Raspberry Pi Zero W Rev 1.1

CPU temp: 46.54'C
GPU temp: 46.5'C

Throttled flags: 0x0

Throttled now:  no
Throttled past: no

Undervoltage now:  no
Undervoltage past: no

Frequency capped now:  no
Frequency capped past: no
[root@pikvm ~]#
````

opiedopy, 2021-08-11:
> I put Morioeye OS on it, but my camera was on order. I meant to have
> turned it off, but forgot. It was sitting alone in free air with just the
> USB power on it.  I am thinking to avoid the PiZeroW in the future and mark
> this one uo to the school of hard knocks.  So many others seem to find the
> PiZero dead after a few weeks. Bummer.

I did an analysis of the Pi Zero failure.  I traced the problem to a $7.99
eBay power supply sold specifically for Raspberry Pi but it put out 6.1
volts.  The poor Pi Zero lasted 14 days anyway, but it was my fault for
buying a craxxy power supply and not checking it.  I am sorry that I blamed
the Pi Zero!

mdevaev, 2021-08-12:
It seems to me that from all this we can conclude that the OS does not kill Zero. Either an error in the circuit diagram, or an excess of the supply voltage, or some other external factors. RNGD did cause heating, however, 60 degrees due to the load is not the same as 60 degrees due to overvoltage. Because some parts of the chip are designed for overload. Hovered, I solved the problem with RNGD and released an update that forcibly applies the settings for this.

Given that the cases of dead Zero are small compared to the number of Pi-KVM installations, I think this issue can be considered resolved. I recommend checking the circuits and the power supply voltage to avoid such sad situations in the future.

If someone does not agree, I will re-open this issue. Also, if you want to send us your dead Zero for analysis, contact me via discord.

closed 2021-08-12 by mdevaev