pikvm/pikvm #184: Ethernet gadget

sharicov, 2021-02-14:
Could you please enable Ethernet gadget on the raspberry as described in following topic:
https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/ethernet-gadget
Tried to do it manually after writing the image on the SD, didn't work.
Would be extremely usable for maintenance activities (like changing WiFi credentials without removing the SD card.
Raspberry is recognized as Ethernet device by the host computer, IP can be provided by host computer by enabling internet sharing of host computer (WiFi or Ethernet).
Can be also useful to access PI_KVM over host computer's network (use case Raspberry PI zero and 5Ghz only wifi,or Ethernet available.
Thanks in advance!

mdevaev, 2021-02-14:
This feature is already described in the documentation, and the link is on the main page.
https://github.com/pikvm/pikvm/blob/master/pages/cookbook.md#enable-ethernet-over-usb-network

sharicov, 2021-02-15:
Sorry, didn't see it at first.
Tried following it and couldn't make it work under windows host.
Appears as CDC-ECM in the device manager - couldn't find windows 10 driver for this one.
Tried setting rndis in /etc/kvmd/override.yaml 
starting/stopping kvmd-otgnet, then rebooting - still CDC-ECM appears in the device manager - no rndis device
Please help

sharicov, 2021-02-15:
On the same topic, when trying to connect to Pi-KVM over USB (ECM) I am not able to connect over SSH.
Port 20 is opened in otgnet: firewall:  allow_tcp: [20, 80, 443]
kvmd-otgnet stop/start
ping works 80,443 and 67 ports are opened can access with "telnet IP PORT", however port 20 is still inaccessible.
Please advise

mdevaev, 2021-02-16:
Ssh uses port 22.

mdevaev, 2021-02-21:
Also see here: https://github.com/pikvm/pikvm/blob/master/pages/usb_ethernet.md
Available in the latest version.

sharicov, 2021-02-25:
> Ssh uses port 22.

Was a typo, checked with 22, didn't work (nor ssh, nor telnet 22)

mdevaev, 2021-02-25:
Show the entire override.yaml

sharicov, 2021-02-25:
#                                                                  #
# All overridden parameters will be applied AFTER other configs    #
# and "!include" directives and BEFORE validation.                 #
# Not: Sections should be combined under shared keys.              #
#                                                                  #
####################################################################
#
#########
# Wrong #
#########
#kvmd:
#    gpio:
#        drivers: ...
#kvmd:
#    gpio:
#        scheme: ...
#
###########
# Correct #
###########
#kvmd:
#    gpio:
#        drivers: ...
#        scheme: ...
#
###########
# Example #
###########
#vnc:
#    # See https://github.com/pikvm/pikvm/blob/master/pages/vnc.md
#    keymap: /usr/share/kvmd/keymaps/ru # Set russian keymap
#    auth:
#        vncauth:
#            enabled: true # Enable auth via /etc/kvmd/vncpasswd
otg:
    devices:
        ethernet:
            enabled: true
            driver: ecm
            host_mac: 48:6f:73:74:50:43
            kvm_mac: 42:61:64:55:53:42
otgnet:
    firewall:
        allow_tcp: [22, 80, 443]
otg:
    devices:
        drives:
            enabled: true  # Set it to true to enable
            count: 1  # +1 drive, default value
            default:  # Default configuration for the all extra drives
                cdrom: false  # Default value (false for the generic flash drive)
                rw: false # Read-only by default

mdevaev, 2021-02-25:
Try to manual restart sshd after boot

sharicov, 2021-02-25:
Just to make clear, I am able to access ssh to PI-KVM from the host computer over WI-FI, what is not working is the ssh connection from the pc connected to PI-KVM over usb (ethernet over usb).
Updated the PI-KVM today, now getting following error:
kvmd-otgnet start
kvmd.apps.otgnet                  INFO --- Using OTG gadget 'kvmd' ...
Traceback (most recent call last):
  File "/usr/bin/kvmd-otgnet", line 9, in <module>
    main()
  File "/usr/lib/python3.9/site-packages/kvmd/apps/otgnet/__init__.py", line 212, in main
    options.cmd()
  File "/usr/lib/python3.9/site-packages/kvmd/apps/otgnet/__init__.py", line 94, in start
    asyncio.run(self.__run(True))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/lib/python3.9/site-packages/kvmd/apps/otgnet/__init__.py", line 100, in __run
    netcfg = self.__make_netcfg()
  File "/usr/lib/python3.9/site-packages/kvmd/apps/otgnet/__init__.py", line 145, in __make_netcfg
    iface = self.__find_iface()
  File "/usr/lib/python3.9/site-packages/kvmd/apps/otgnet/__init__.py", line 182, in __find_iface
    with open(path) as iface_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/config/usb_gadget/kvmd/functions/ecm.usb0/ifname'

sharicov, 2021-02-25:
Seems like after update g_ether completely stop working.
Don't see usb0 network device (ip a) both with ecm and rndis

mdevaev, 2021-02-25:
`journalctl -u kvmd-otg`

mdevaev, 2021-07-04:
As far as I remember, I answered you in Discord, so I'm closing this

closed 2021-07-04 by mdevaev