pikvm/pikvm #185: Informative console output over HDMI

sharicov, 2021-02-14:
Could you please show some useful output via Raspberry's HDMI for debugging purposes.
I cannot connect WIFI for some reason, would like to see some info when connecting Raspberry to monitor via HDMI.
For example dmesg or tail -f /ver/log/messages
Thanks in advance

alexsalex, 2021-02-14:
Please, check this:
https://github.com/pikvm/pikvm/issues/181

mdevaev, 2021-02-14:
Use the keyboard and the raspberry's HDMI output. The usual diagnostic tools will work as described above.

closed 2021-02-14 by mdevaev

sharicov, 2021-02-15:
> Use the keyboard and the raspberry's HDMI output. The usual diagnostic tools will work as described above.
Thanks for quick responce,
The problem is, keyboard didn't work for me when connected it directly to my raspberry zero W's USB (tried two working keyboards). Very strange.

prepperdev, 2021-02-15:
Just my 2 cents: Raspberry Pi Zero W is an extremely flaky device. It costs less than a "real" Pi, but then it keeps charging you by being uncooperative.

sharicov, 2021-02-15:
It has it's advantages though, consuming less battery, much more compact, disposable for it's cost...

mdevaev, 2021-02-15:
For zero, you will have to disable usb otg to make the keyboard work. You can also buy a USB UART adapter and use it, it is a reliable solution.

But I recommend using pi4. I can't physically create a miracle for you from one USB controller and one core.

sharicov, 2021-02-15:
Understood, thanks!
How do I disable OTG?
Also brings me back to issue 187 I had open (https://github.com/pikvm/pikvm/issues/187) how can I connect bluetooth keyboard (mouse)?

mdevaev, 2021-02-15:
Edit `/boot/config.txt` on the first sd card partition and comment line with `dwc2`. Just keep in mind that after this, the web interface will stop working.

sharicov, 2021-02-15:
Thank you very much, you are doing a great job with this project!